YosysHQ/yosys #1962: ABC: Assertion failed - Gia_ManInsertMfs

kost, 2020-04-18:
## Steps to reproduce the issue

Building SaxonSoc available here:
https://github.com/SpinalHDL/SaxonSoc

With instructions documented here:
https://github.com/dok3r/ulx3s-saxonsoc/blob/master/scripts/build.sh

Using following yosys command:
```
synth_ecp5 -abc9 -json bin/toplevel.json
```
Getting following error:
```
4.49.9.4. Executing ABC9_EXE pass (technology mapping using ABC9).

4.49.9.5. Executing ABC9.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_lut <abc-temp-dir>/input.lut
ABC: + read_box <abc-temp-dir>/input.box
ABC: + &read <abc-temp-dir>/input.xaig
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =   6349/   7794  and =   38507  lev =   76 (6.38)  mem = 0.80 MB  box = 761  bb = 171
ABC: + &scorr
ABC: Warning: The network is combinational.
ABC: + &sweep
ABC: + &dc2
ABC: + &dch -f
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =   6349/   7794  and =   51288  lev =   32 (4.32)  mem = 0.94 MB  ch = 7296  box = 735  bb = 171
ABC: + &if -W 200 -v
ABC: K = 7. Memory (bytes): Truth =    0. Cut =   60. Obj =  140. Set =  636. CutMin = no
ABC: Node =   51288.  Ch =  6650.  Total mem =   11.05 MB. Peak cut mem =    0.43 MB.
ABC: P:  Del = 4893.00.  Ar =   43144.0.  Edge =    56334.  Cut =   628766.  T =     0.28 sec
ABC: P:  Del = 4893.00.  Ar =   40562.0.  Edge =    53770.  Cut =   600816.  T =     0.29 sec
ABC: P:  Del = 4893.00.  Ar =   17779.0.  Edge =    40722.  Cut =  1336254.  T =     0.56 sec
ABC: F:  Del = 4893.00.  Ar =   12985.0.  Edge =    37536.  Cut =  1227765.  T =     0.55 sec
ABC: A:  Del = 4892.00.  Ar =   12337.0.  Edge =    34716.  Cut =  1200384.  T =     0.87 sec
ABC: A:  Del = 4892.00.  Ar =   12274.0.  Edge =    34569.  Cut =  1217243.  T =     0.74 sec
ABC: Total time =     3.28 sec
ABC: + &mfs
ABC: Assertion failed: iLitNew >= 0 (src/aig/gia/giaMfs.c: Gia_ManInsertMfs: 388)
ABC: Aborted (core dumped)
ERROR: ABC: execution of command "/opt/ulx3s/bin/yosys-abc -s -f /tmp/yosys-abc-pinohi/abc.script 2>&1" failed: return code 134.
make: *** [makefile:41: bin/toplevel.json] Error 1
```
Last known yosys version which works (from somewhere around 2020-03-13):
```
root@260d3effb874:/# yosys -V
Yosys 0.9+1706 (git sha1 c0a00913, g++ 8.3.0 -Os) 
```

Last known yosys version which failed to build (from somewhere around 2020-03-19):
```
root@f2ba9151df85:/# yosys -V
Yosys 0.9+1706 (git sha1 0cbf1023, g++ 8.3.0 -Os)
```
So, code changes between 2020-03-13 c0a00913 and 2020-03-19 0cbf1023 introduced the problem.

Note that I tried on latest master version of yosys and got same error.

## Expected behavior

That yosys does not crash with ABC: Aborted (core dumped).

## Actual behavior

Yosys crashes with ABC: Aborted (core dumped).

ZirconiumX, 2020-04-18:
#1459 was the same bug, which was fixed upstream with 37537609. It's quite odd for it to be back again, but likely means the fix didn't actually fix it.

daveshah1, 2020-04-18:
Yes, based on past experience of this bug it may well not be a change in those Yosys revisions that caused it per se, but rather unrelated changes to the netlist resulting from those commits triggering the never-properly-fixed bug in ABC9.

daveshah1, 2020-04-18:
Sorry, closed by a mistake.

eddiehung, 2020-04-20:
Yes, this is another known issue in ABC. It looks like I'll have to disable "&mfs" again.

eddiehung, 2020-04-20:
@kost: please give #1974 a try!

kost, 2020-04-21:
Just tested and can confirm it works.

I have used 215ac18f736cff06a4186e11b9abaa3d669f32e7 commit hash of dev branch of https://github.com/SpinalHDL/SaxonSoc.git which is known to cause this issue.

eddiehung, 2020-04-23:
Closed by #1974

closed 2020-04-23 by eddiehung