YosysHQ/yosys #1965: yosys: error while loading shared libraries: libffi.so.6: cannot open shared object file: No such file or director

jchidley, 2020-04-18:
## Steps to reproduce the issue

*Provide instructions for reproducing the issue. Make sure to include
all necessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

run 

```bash
makepkg -si
``` 
on Arch linux for [AUR yosys-git](https://aur.archlinux.org/packages/yosys-git/), then run:

```bash
yosys
```

See https://stackoverflow.com/help/mcve for some information on how to
create a Minimal, Complete, and Verifiable example (MCVE).

Please do not waste our time with issues that lack sufficient infohttps://aur.archlinux.org/packages/yosys-git/rmation
to reproduce the issue easily. We will simply close those issues.

Contact https://www.symbioticeda.com/ if you need commercial support for Yosys.

## Expected behavior

Yosys runs normally

## Actual behavior

```bash
yosys: error while loading shared libraries: libffi.so.6: cannot open shared object file: No such file or director
```

Mitigation: 

```bash
ln -sf /usr/lib/libffi.so.7 /usr/lib/libffi.so.6
```

daveshah1, 2020-04-18:
I attempted to reproduce and get the following error in `make test`:

<pre>Running plugin.sh..
<b>plugin.cc:1:10:</b> <font color="#CC0000"><b>fatal error: </b></font>kernel/rtlil.h: No such file or directory
    1 | #include <font color="#CC0000"><b>&quot;kernel/rtlil.h&quot;</b></font>
      |          <font color="#CC0000"><b>^~~~~~~~~~~~~~~~</b></font>
compilation terminated.</pre>

So there is another bug here; but I suspect the cause of your bug is due to having out of date files in your build directory.

jchidley, 2020-04-18:
I wondered about out-of-date stuff too.  So I did a full Arch upgrade, deleted the entire yosys-git (rm -r yosys-git) directory and reinstalled from scratch.  No change.  It might be a problem with libdiff itself

Xiretza, 2020-04-18:
The failing tests have already  been mentioned in https://github.com/YosysHQ/yosys/commit/f11dd6e20838bce1b1fa8e085570d2126484d10f#r38433102, I think I'll open a separate issue for that though.

Xiretza, 2020-04-18:
Can't reproduce, the yosys binary in the `yosys-git` package I just built (in a clean chroot) links against `libffi.so.7`:

```
$ ldd yosys 
	linux-vdso.so.1 (0x00007ffcf9fe9000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f938f294000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f938f14e000)
	libpython3.8.so.1.0 => /usr/lib/libpython3.8.so.1.0 (0x00007f938ed8e000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f938ed6c000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f938ed67000)
	libboost_python38.so.1.72.0 => /usr/lib/libboost_python38.so.1.72.0 (0x00007f938ed27000)
	libboost_filesystem.so.1.72.0 => /usr/lib/libboost_filesystem.so.1.72.0 (0x00007f938ed05000)
	libreadline.so.8 => /usr/lib/libreadline.so.8 (0x00007f938ecb3000)
	libffi.so.7 => /usr/lib/libffi.so.7 (0x00007f938eca7000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007f938ec8d000)
	libtcl8.6.so => /usr/lib/libtcl8.6.so (0x00007f938ead1000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f938eab7000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f938e8ef000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007f93900b9000)
	libutil.so.1 => /usr/lib/libutil.so.1 (0x00007f938e8ea000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007f938e8df000)
	libncursesw.so.6 => /usr/lib/libncursesw.so.6 (0x00007f938e86e000)
```

And this is surely out of scope anyway, if you can't reproduce it with a yosys built directly from the git tree, it's a package issue, not a yosys issue.

FFY00, 2020-04-19:
If you build yosys-git against a specific libffi version, when libffi updates you will need to rebuild it.

Everything here is working as expected, you just need to rebuild yosys-git against the new libffi version.

And as @Xiretza suggested, you should build the package in a clean chroot (use `extra-x86_64-build` from `devtools`).

closed 2020-04-19 by daveshah1