YosysHQ/yosys #1966: tests: various/plugin.sh fails when PREFIX != /usr/local/share

Xiretza, 2020-04-18:
This has already been brought up in https://github.com/YosysHQ/yosys/commit/f11dd6e20838bce1b1fa8e085570d2126484d10f#r38433102, but I'll add it here for better visibility and tracking.

## Steps to reproduce the issue

```
make -j8 PREFIX=/usr/share test
```

## Expected behavior

The tests complete without error.

## Actual behavior

`tests/various/plugin.cc` fails to compile with:

```
plugin.cc:1:10: fatal error: 'kernel/rtlil.h' file not found
#include "kernel/rtlil.h"
         ^~~~~~~~~~~~~~~~
```

This is due to the replacement of include path flags that happens in `plugin.sh`, which assumes `PREFIX=/usr/local/share`.

eddiehung, 2020-04-18:
Sorry, dropped that previous comment. Thanks for filing, will look at it next week.

eddiehung, 2020-04-20:
Hi @Xiretza, can you give #1973 a try please?

closed 2020-04-22 by eddiehung

aimylios, 2020-05-01:
Unfortunately #1973 has not resolved this issue for me. I'm building nightly yosys RPMs for Fedora, and I get the following error when running the tests:
```
Running plugin.sh..
plugin.cc:1:10: fatal error: kernel/rtlil.h: No such file or directory
    1 | #include "kernel/rtlil.h"
      |          ^~~~~~~~~~~~~~~~
```

I assume this is because the include folder is not specified correctly. `DATDIR` is set to `/usr/share/yosys` (as expected), but in Fedora the include files are located in `/usr/include/yosys`. Consequently, the path substitution `CXXFLAGS=${CXXFLAGS//$DATDIR/..\/..\/share}` does not work (there is no occurence of `DATDIR` in `CXXFLAGS`).

I'm now using the following code in plugin.sh to replace the include path, but there must be a less hacky and more reliable solution to force the local source files to be included.
```
INCDIR=${CXXFLAGS#* -I}
INCDIR=${INCDIR%% -*}
INCDIR=${INCDIR//\//\\\/}
CXXFLAGS=${CXXFLAGS//$INCDIR/..\/..}
```

Xiretza, 2020-05-01:
How does fedora build its yosys package? From what I can tell that sort of setup isn't supported by the makefile, headers are [always expected to be in `$(DATDIR)/include`](https://github.com/YosysHQ/yosys/blob/bc380b0b56e0fa269a0d5844280cc4419d438d3a/Makefile#L668). If fedora manually patches the makefile or moves the files around after `make install`, it would be their obligation to make sure everything else works as well.

aimylios, 2020-05-01:
Yes, indeed, Fedora patches the Makefile (see the [SPEC file for 0.9](https://src.fedoraproject.org/rpms/yosys/blob/master/f/yosys.spec) and the [patch](https://src.fedoraproject.org/rpms/yosys/blob/master/f/yosys-cfginc.patch)). If that's the preferred solution, I can of course add another patch to my nightly builds to fix plugin.sh. I just thought there might be a more universal solution to the problem (like the hacky one I posted above).

PR #1973 (2020-04-22): tests: fix various/plugin.sh when PREFIX != /usr/local/share
Fixes #1966