YosysHQ/yosys #1970: ABC failure due to wrongly formed input.lut temp file

acomodi, 2020-04-20:
## Steps to reproduce the issue

[test.zip](https://github.com/YosysHQ/yosys/files/4503542/test.zip)

The `test.zip` archive contains a test design and the tcl synthesis script.

Reproduce:
1. unzip test.zip
2. run `yosys -p "tcl synth.tcl" basys3_top.v rom.v  srl_init_tester.v`

Yosys version: `Yosys 0.9+2406 (git sha1 2b1fb8c0, clang 6.0.0-1ubuntu2 -fPIC -Os)`

## Expected behavior

*Please describe the behavior you would have expected from the tool.*

No errors produced

## Actual behavior

*Please describe how the behavior you see differs from the expected behavior.*

ABC fails as the `input.lut` temporary file seems to be wrong.
In details, the design contains an explicit `LUT2` instance, which seems to be replicated in the `input.lut` file as follows:

```
# LUT1
1 1 127
# LUT2
2 2 127 238
# $paramod\LUT2\INIT=4'1100
2 2 127 238
# LUT3
3 3 127 238 407
# LUT4
4 3 127 238 407 472
# LUT5
5 3 127 238 407 472 631
# LUT6
6 5 127 238 407 472 631 642
```

The LUT2 is instantiated in `basys3_top.v`.

eddiehung, 2020-04-20:
In the future, you want to make it easy as possible for someone to fix this for you so please put some effort into reducing this testcase. The `srl_init_tester.v` and `rom.v` files are not necessary here, and `basys3_top.v` can be reduced significantly. This not only makes it easier to identify the problem, but can also be used as a testcase so that it doesn't get broken in the future.

acomodi, 2020-04-20:
@eddiehung Sure, you are right. Actually the only thing that is necessary in the design is the LUT2 instantiation. I have reduced the top.v file with only that.

Here is the new zip:
[test.zip](https://github.com/YosysHQ/yosys/files/4504705/test.zip)

And the command to reproduce it:
`yosys -p "tcl synth.tcl" top.v`

eddiehung, 2020-04-20:
Thanks @acomodi, doing the same would be very helpful in the future and encourages a prompt fix. Please give #1972 a try!

acomodi, 2020-04-20:
@eddiehung Tested the initial design and works! Thanks

PR #1972 (2020-04-20): abc9_ops: -prep_lut to be more robust
Fixes #1970
closed 2020-04-20 by eddiehung