YosysHQ/yosys #1980: Verific unconditionally prints license information to stdout

whitequark, 2020-04-22:
This currently breaks nMigen, as [described here](https://github.com/nmigen/nmigen/pull/364). It really should use stderr instead to keep Yosys usable in *nix pipelines.

daveshah1, 2020-04-22:
This only seems to occur when linked with evaluation Verific, a Symbiotic EDA build with proper Verific seems fine.

whitequark, 2020-04-22:
cc @hofstee 

@daveshah1 Still, can it be fixed for evaluation Verific too? I can't think of any reason it should use stdout and not stderr.

cliffordwolf, 2020-04-22:
> can it be fixed for evaluation Verific too?

Suppressing that message would be a violation of the Verific evaluation license afaiu.

whitequark, 2020-04-22:
@cliffordwolf Not suppressing. Getting it to stderr and not to stdout, where it would be *more* visible if anything.

cliffordwolf, 2020-04-22:
Still a violation of the eval license. Afair this could only be done by patching the verific eval binary, as they are printing this in a global constructor or something. It's not something that is under the control of the application using the library.

whitequark, 2020-04-22:
You are [entirely correct about the global constructor](https://github.com/nmigen/nmigen/pull/364#issuecomment-617748888) but I didn't realize there is no way around except to patch it. That's quite obnoxious.

closed 2020-04-22 by whitequark