YosysHQ/yosys #1990: Dynamic bit write with signed expression fails

daveshah1, 2020-04-23:
Minimised from an example by @smunaut 

```verilog
module top(input [7:0] wr_mask, output reg [7:0] mem_dm_w);

	localparam DATA_WIDTH = 32;
	localparam MASK_WIDTH = 8;
	localparam NW = 2;

	always @(*)
	begin : map
		integer n, x, o;

		// Map actual bits
		for (n=0; n<MASK_WIDTH; n=n+1)
		begin
			// Determine position
			x = n % NW;	// Which SPRAM
			o = n / NW;	// Which nibble inside that SPRAM

			// Map IO
			mem_dm_w[( 4*x)+   o    ] = ~wr_mask[n];
		end
	end
endmodule
```
gives
```
ERROR: Found error in internal cell \top.$shl$signed_bitsel.v:0$134 ($shl) at kernel/rtlil.cc:850:
  attribute \src "signed_bitsel.v:0"
  cell $shl $shl$signed_bitsel.v:0$134
    parameter \Y_WIDTH 8
    parameter \B_WIDTH 32
    parameter \A_WIDTH 8
    parameter \B_SIGNED 1
    parameter \A_SIGNED 0
    connect \Y $shl$signed_bitsel.v:0$134_Y
    connect \B $add$signed_bitsel.v:19$133_Y
    connect \A $and$signed_bitsel.v:0$131_Y
  end
```

This appears to be a regression caused by #1851, reverting the merge commit causes it to build fine.

daveshah1, 2020-04-23:
Possible patch, although I don't know if this handles all cases correctly (e.g. if one of the vector bounds is negative, is this even legal?)

```patch
diff --git a/frontends/ast/simplify.cc b/frontends/ast/simplify.cc
index 837c14ad..5e12e3ac 100644
--- a/frontends/ast/simplify.cc
+++ b/frontends/ast/simplify.cc
@@ -1774,7 +1774,7 @@ bool AstNode::simplify(bool const_fold, bool at_zero, bool in_lvalue, int stage,
                AstNode *range = children[0]->children[0];
 
                if (range->children.size() == 1) {
-                       shift_expr = range->children[0]->clone();
+                       shift_expr = new AstNode(AST_TO_UNSIGNED, range->children[0]->clone());
                } else {
                        shift_expr = range->children[1]->clone();
                        AstNode *left_at_zero_ast = range->children[0]->clone();
```

daveshah1, 2020-04-23:
Even with the above patch, I also get a logic loop warning that wasn't shown before:
```
Warning: found logic loop in module top:
    cell $and$signed_bitsel.v:0$100 ($and)
    cell $and$signed_bitsel.v:0$112 ($and)
    cell $and$signed_bitsel.v:0$124 ($and)
    cell $and$signed_bitsel.v:0$136 ($and)
    cell $and$signed_bitsel.v:0$52 ($and)
    cell $and$signed_bitsel.v:0$64 ($and)
    cell $and$signed_bitsel.v:0$76 ($and)
    cell $and$signed_bitsel.v:0$88 ($and)
    cell $or$signed_bitsel.v:0$101 ($or)
    cell $or$signed_bitsel.v:0$113 ($or)
    cell $or$signed_bitsel.v:0$125 ($or)
    cell $or$signed_bitsel.v:0$137 ($or)
    cell $or$signed_bitsel.v:0$53 ($or)
    cell $or$signed_bitsel.v:0$65 ($or)
    cell $or$signed_bitsel.v:0$77 ($or)
    cell $or$signed_bitsel.v:0$89 ($or)
    wire $and$signed_bitsel.v:0$100_Y [0]
    wire $and$signed_bitsel.v:0$112_Y [0]
    wire $and$signed_bitsel.v:0$124_Y [0]
    wire $and$signed_bitsel.v:0$136_Y [0]
    wire $and$signed_bitsel.v:0$52_Y [0]
    wire $and$signed_bitsel.v:0$64_Y [0]
    wire $and$signed_bitsel.v:0$76_Y [0]
    wire $and$signed_bitsel.v:0$88_Y [0]
    wire $or$signed_bitsel.v:0$101_Y [0]
    wire $or$signed_bitsel.v:0$113_Y [0]
    wire $or$signed_bitsel.v:0$125_Y [0]
    wire $or$signed_bitsel.v:0$53_Y [0]
    wire $or$signed_bitsel.v:0$65_Y [0]
    wire $or$signed_bitsel.v:0$77_Y [0]
    wire $or$signed_bitsel.v:0$89_Y [0]
    wire \mem_dm_w [0]
Warning: found logic loop in module top:
    cell $and$signed_bitsel.v:0$100 ($and)
    cell $and$signed_bitsel.v:0$112 ($and)
    cell $and$signed_bitsel.v:0$124 ($and)
    cell $and$signed_bitsel.v:0$136 ($and)
    cell $and$signed_bitsel.v:0$52 ($and)
    cell $and$signed_bitsel.v:0$64 ($and)
    cell $and$signed_bitsel.v:0$76 ($and)
    cell $and$signed_bitsel.v:0$88 ($and)
    cell $or$signed_bitsel.v:0$101 ($or)
    cell $or$signed_bitsel.v:0$113 ($or)
    cell $or$signed_bitsel.v:0$125 ($or)
    cell $or$signed_bitsel.v:0$137 ($or)
    cell $or$signed_bitsel.v:0$53 ($or)
    cell $or$signed_bitsel.v:0$65 ($or)
    cell $or$signed_bitsel.v:0$77 ($or)
    cell $or$signed_bitsel.v:0$89 ($or)
    wire $and$signed_bitsel.v:0$100_Y [1]
    wire $and$signed_bitsel.v:0$112_Y [0]
    wire $and$signed_bitsel.v:0$124_Y [0]
    wire $and$signed_bitsel.v:0$136_Y [0]
    wire $and$signed_bitsel.v:0$52_Y [0]
    wire $and$signed_bitsel.v:0$64_Y [0]
    wire $and$signed_bitsel.v:0$76_Y [0]
    wire $and$signed_bitsel.v:0$88_Y [0]
    wire $or$signed_bitsel.v:0$101_Y [0]
    wire $or$signed_bitsel.v:0$113_Y [0]
    wire $or$signed_bitsel.v:0$125_Y [0]
    wire $or$signed_bitsel.v:0$53_Y [0]
    wire $or$signed_bitsel.v:0$65_Y [0]
    wire $or$signed_bitsel.v:0$77_Y [0]
    wire $or$signed_bitsel.v:0$89_Y [0]
    wire \mem_dm_w [0]
Warning: found logic loop in module top:
    cell $and$signed_bitsel.v:0$100 ($and)
[... 172 lines cut ...]
    cell $or$signed_bitsel.v:0$101 ($or)
    cell $or$signed_bitsel.v:0$113 ($or)
    cell $or$signed_bitsel.v:0$125 ($or)
    cell $or$signed_bitsel.v:0$137 ($or)
    cell $or$signed_bitsel.v:0$53 ($or)
    cell $or$signed_bitsel.v:0$65 ($or)
    cell $or$signed_bitsel.v:0$77 ($or)
    cell $or$signed_bitsel.v:0$89 ($or)
    wire $and$signed_bitsel.v:0$100_Y [7]
    wire $and$signed_bitsel.v:0$112_Y [0]
    wire $and$signed_bitsel.v:0$124_Y [0]
    wire $and$signed_bitsel.v:0$136_Y [0]
    wire $and$signed_bitsel.v:0$52_Y [0]
    wire $and$signed_bitsel.v:0$64_Y [0]
    wire $and$signed_bitsel.v:0$76_Y [0]
    wire $and$signed_bitsel.v:0$88_Y [0]
    wire $or$signed_bitsel.v:0$101_Y [0]
    wire $or$signed_bitsel.v:0$113_Y [0]
    wire $or$signed_bitsel.v:0$125_Y [0]
    wire $or$signed_bitsel.v:0$53_Y [0]
    wire $or$signed_bitsel.v:0$65_Y [0]
    wire $or$signed_bitsel.v:0$77_Y [0]
    wire $or$signed_bitsel.v:0$89_Y [0]
    wire \mem_dm_w [0]
```

cliffordwolf, 2020-04-27:
Here is an improved minimal example:

```
module top (
        input [31:0] din,
        input signed [4:0] n,
        output reg [31:0] dout
);
        always @(*) begin
                // avoid logic loop issue
                dout = 0;

                // this version fails
                // dout[n^1] = din[n];

                // this version works
                dout[n^1 +: 1] = din[n];
        end
endmodule
```

There are two issues.

First is the issue that this code (without the added default assignment) infers a latch. We should detect that and then fall back to the original "big case" method of implementing. As a first work-around I will add a Verilog attribute that can be set on a register to force the "big case" method.

Second we have an issue with signed variables. I'm investigating this right now. The patch proposed by @daveshah1 is a good start but doesn't seem to do the right thing in all cases according to my preliminary tests.

smunaut, 2020-04-27:
How does the original minimized example by @daveshah1 describes a latch ?  All 8 bits of mem_dm_w are always assigned ... And which bits of mem_dm_w  is assigned from which bits of wr_mask is known at compile time, it's in no way dependent on the input signals.

Also, even though I'm using `integer` the value is actually never negative, not sure if that matters or not ...

cliffordwolf, 2020-05-08:
The signed expression aspect of this issue has been fixed in #2005. Issue #2040 replaces this one for the logic loop problem.

commit bbbce0d1 (2020-05-02): Add "nowrshmsk" attribute, fix shift-and-mask bit slice write for signed offset, fixes #1990
closed 2020-05-08 by cliffordwolf