YosysHQ/yosys #1991: Default module input ports not handled properly

ZipCPU, 2020-04-23:
## Steps to reproduce the issue

Using the [test.zip](https://github.com/YosysHQ/yosys/files/4523054/test.zip) file below, containing both a verilog file and a SymbiYosys script, run SymbiYosys.

The key line at issue in the file is the input line of the submodule,
```
    input wire i_stall = 0;
```

If I understand how this is to work properly, `i_stall` should be set to zero only if it is not connected to any outside port.  Two tests are contained within the file--one that leaves this pin unconnected for which I expect the default to hold, and a second one that connects this pin to a top-level input for which I expect the default to be ignored.

Within the file, there are two properties.  The first checks that, if `i_stall` is left at its default, then the output should match a similar output that isn't constrained by `i_stall`.  The second property, a `cover()` statement, checks that `i_stall` can still be non-zero if included in a top level module input.

## Expected behavior

I expect both properties to pass--the assertion in the induction proof and the cover property when running cover.

## Actual behavior

When using Yosys from today's build, both properties fail.

When using the Symbiotic EDA Suite (commercial version) dated 2 March, 2020, the assertion succeeds bu the cover property fails.

Dan

whitequark, 2020-04-23:
> If I understand how this is to work properly, `i_stall` should be set to zero only if it is not connected to any outside port.

As far as I know, the code you wrote, in Verilog, implements a "continuous net declaration assignment" (IEEE 1364-2005 §6.1.1), and therefore if you connect anything to the input of the port, you get a driver conflict, which results in a `'x` value. I cannot find anything about "default input ports" in the LRM. Do you happen to know where in the LRM they are defined?

cliffordwolf, 2020-04-23:
It's a SystemVerilog feature. From IEEE Std 1800-2012:

[image]

I think the problem in that example is that it is not using ANSI-style port declarations.

ZipCPU, 2020-04-23:
> I think the problem in that example is that it is not using ANSI-style port declarations

This may well be.  If I switch to ANSI port declarations, the problem vanishes.  Further, the spec did say "Defaults can be specified only for inputs ports and *only in*ANSI style declarations.*"

If we go with the possibility that the test case is at fault, perhaps the "right" answer should be to create a warning for this case.

ZipCPU, 2020-04-23:
Sorry ... I didn't mean to close it just now--I hit the wrong button.  (Oops!)

cliffordwolf, 2020-04-23:
> If we go with the possibility that the test case is at fault, perhaps the "right" answer should be to create a warning for this case.

There already is a warning:

```
VERIFIC-WARNING [VDB-1003] imcve.v:46: net 'i_stall' is already driven by input port 'i_stall'
```

closed 2020-04-23 by cliffordwolf