YosysHQ/yosys #2003: opt_expr prevents driver/constant conflict warning

daveshah1, 2020-04-25:
```verilog
module top(input [3:0] a, output [3:0] y);
	wire [3:0] x = a + 1;
	wire [3:0] tmp = 0;
	assign tmp = x;
	assign y = tmp | x;
endmodule
```

Running `yosys -qp "opt_clean" driver_constant.v` gives
```
Warning: Driver-driver conflict for $add$driver_constant.v:2$1_Y [3] between cell $add$driver_constant.v:2$1.Y and constant 1'0 in top: Resolved using constant.
Warning: Driver-driver conflict for $add$driver_constant.v:2$1_Y [2] between cell $add$driver_constant.v:2$1.Y and constant 1'0 in top: Resolved using constant.
Warning: Driver-driver conflict for $add$driver_constant.v:2$1_Y [1] between cell $add$driver_constant.v:2$1.Y and constant 1'0 in top: Resolved using constant.
Warning: Driver-driver conflict for $add$driver_constant.v:2$1_Y [0] between cell $add$driver_constant.v:2$1.Y and constant 1'0 in top: Resolved using constant.
```

Running `yosys -qp "opt_expr; opt_clean" driver_constant.v` gives no warnings.

It seems wrong for something that has the possibility to cause significant unexpected behaviour (and arguably should be an error), to not even produce a warning depending on the sequence of passes called.

nakengelhardt, 2020-08-10:
It's not really either `opt_expr` or `opt_clean`'s job to find things like this in the design. `opt_expr` just looks at the immediate local context and so doesn't see the issue, `opt_clean` happens to spot it but doesn't know what to do about it, so gives a warning. IMO the real issue here is that `check` reports 0 problems with this design!

mwkmwkmwk, 2020-08-10:
Seconded.  I would also add that perhaps `check` should be called earlier in the synth flows. — currently it's called only after `opt_expr` and `opt_clean`, which has a good chance of transforming a "bad" design into a "good" one.

widlarizer, 2026-08-24:
6. Executing CHECK pass (checking for obvious problems).
Checking module top...
Warning: Drivers conflicting with a constant 1'0 driver:
    port Y[0] of cell $add$g/check.v:2$1 ($add)
    port Y[1] of cell $add$g/check.v:2$1 ($add)
    port Y[2] of cell $add$g/check.v:2$1 ($add)
    port Y[3] of cell $add$g/check.v:2$1 ($add)

closed 2026-08-24 by widlarizer