YosysHQ/yosys #2009: Fail to parse hexadecimal constants that start with a B

jshaker000, 2020-04-30:
As the topic says,

```
 1. Executing Verilog-2005 frontend: Instruction_Decoder.v
Parsing Verilog input from `Instruction_Decoder.v' to AST representation.
Instruction_Decoder.v:157: ERROR: syntax error, unexpected TOK_ID   
```

The offending line (was)

`                   i_instruction == 'hb ? // JIO - jump to ADDR if last ALU op was odd`
Changing to

`                   i_instruction == 4'h0b ? // JIO - jump to ADDR if last ALU op was odd`

Fixed the problem

I assuem yosys was unsure whether to interpret it as a constant 0b1011 or to look for a binary number or ?

daveshah1, 2020-04-30:
I can't reproduce this. Is your Yosys up to date?

dh73, 2020-04-30:
I can't reproduce it either:
``` 
Yosys 0.9+2406 (git sha1 c944d844, clang 8.0.1 -fPIC -Os)

yosys> read_verilog  example.v 
1. Executing Verilog-2005 frontend: example.v
Parsing Verilog input from `example.v' to AST representation.
Generating RTLIL representation for module `\test'.
Successfully finished Verilog frontend.
```

But since I don't have a mac machine to run it, and that `UNKNOWN` in place of the sha number looks suspicious, it might be something related to the OS/package versions. 

Maybe @mmicko or @nakengelhardt are the best persons to do a quick check?.

whitequark, 2020-04-30:
This was possibly fixed by 5e94bf029179257cb3539cc5096113a139e37ff5.

anishathalye, 2020-05-05:
I'm using Yosys on a macOS machine. The example works for me on tip of tree (584780d776c92bc91731dbc2710dd8d9a624dc70). The [Homebrew version](https://github.com/Homebrew/homebrew-core/blob/master/Formula/yosys.rb) is quite out of date because the latest release of Yosys is 0.9, released August 2019.

@jshaker000 if you build from source manually / install with `brew install --HEAD yosys`, your example should work.

jshaker000, 2020-05-06:
You are right, I think that in this case simply the brew package is quite out of date (actually the arch linux package is quite out of date too - what I use in my VM), but that wouldnt be a yosys bug itself.

closed 2020-05-06 by jshaker000