YosysHQ/yosys #2021: Condition without statement syntax error

towoe, 2020-05-04:
## Steps to reproduce the issue

```verilog
module empty ();
	parameter on = 0;
	if (on);
endmodule
```

## Expected behavior

I am not completely sure if this is actually a valid Verilog syntax, but from the spec I read the following: "if (expression) statement_or_null" so I am assuming that with the "null" part is should be legal to have a condition without a statement.

## Actual behavior

1. Executing Verilog-2005 frontend: empty.v
Parsing Verilog input from `empty.v' to AST representation.
empty.v:3: ERROR: syntax error, unexpected ';'

Sorry if this is just a question and not a bug report and thanks for the support!

dh73, 2020-05-04:
Hello, 
In the IEEE 1800-2015,  `A.6.6 Conditional statements`, shows the same syntax as you pointed out here. In the `if (on);` statement, the `;` is the null construct. However, the example is not correct, since conditional expressions can be used either in procedural assignments or generated instantiations (generate if). In that example, conditional statement is not under any of those.

This is correct, Yosys generates an empty module as expected:
```Verilog
module test ();
        localparam y = 1;
        always @(*) 
                if (y) ;
endmodule
```
Another example with a case statement:
```Verilog
module test (input a, output y);
        always @(*)
                case (a)
                        1'b0: y = 1'b1;
                        default: ;
                endcase
endmodule
```
Your example can be modeled under a `generate-if` construct. Although it will be an empty module instantiating no cell.

nakengelhardt, 2020-05-04:
Tangentially, `statement_or_null` is defined in A.6.4 as `statement | { attribute_instance } ;` but if I try to add the optional attribute it doesn't work:
```
module test ();
	localparam y = 1;
	always @(*)
		if (y) (* foo *) ;
endmodule
```
```
1. Executing Verilog-2005 frontend: stmt.v
Parsing Verilog input from `stmt.v' to AST representation.
stmt.v:4: ERROR: syntax error, unexpected ';'
```
Am I doing it wrong, or is there some restriction on attributes I didn't notice?

towoe, 2020-05-04:
Thanks @dh73 for the detailed reply!

The first example does work for me.

The example I posted was actually just an excerpt of something else. So my minimal example was not correct, but thanks for clarifying what is possible and correct.

But the following with the `generate` does not work as well. @dh73 should it?

```
module empty ();
	localparam on = 1;
	generate
		if (on);
	endgenerate
endmodule
```

The following does work
```
module empty ();
	localparam on = 1;
	generate
		if (on) begin
		end
	endgenerate
endmodule
```

dh73, 2020-05-04:
@nakengelhardt I believe your example is correct, and the first generate example from @towoe it is also correct. What I am not sure if there is a user case to support the optional attribute, nor the conditional generate with no instantiation.

Both are supported by Symbiotic EDA frontend.

towoe, 2020-05-04:
The Verilog code is the output of SystemVerilog to Verilog processing where some asserts (https://github.com/lowRISC/ibex/blob/master/rtl/ibex_id_stage.sv#L970) are not defined and therefore without any instantiations.

Just so I understand it @dh73, the first example is correct and should be accepted by the Verilog frontend? If so I could have a look at the code and do a PR.
And by Symbiotic EDA frontend you mean the SystemVerilog / Verific frontend?

dh73, 2020-05-04:
One second, the RTL in that link uses assertions constructs that are not defined in IEEE 1364-2005 but in IEEE 1800-20xx SystemVerilog,  and therefore, not supported by the Verilog frontend of Yosys. Suffix implication is not supported in Yosys Verilog frontend but Verific frontend.

Pardon me If I am not understanding, but if some tool is processing the SystemVerilog, to generate a Verilog from ibex, it should work with those constructs that are not synthesizable by the target standard and/or does not have a direct translation (such as using compiler directives to encapsulate them out of the compilation scope).

My point is, inline assertions should not be part of a correctly generated Verilog module in the first place, since those constructs still targets SystemVerilog, therefore no `statement_or_null` applies here (altough it is demostrated that Yosys correctly process it).

In the other hand, both optional attribute and generate-if with null statement are valid in IEEE 1364-2005. Verilog frontend enhancements and corner cases synthesis examples that Yosys does wrong, are very welcome by the team.

towoe, 2020-05-05:
You are correct @dh73 that's why the assertions do not end up in the Verilog source and the generate if construct contains `null`.

The tool I use is sv2v by the way. And it seems like @zachjs already did a PR.
Thanks to everyone for jumping on this issue.

nakengelhardt, 2020-05-07:
The null statement issue was fixed in #2028; I've opened a new issue (#2037) for the attribute thing.

closed 2020-05-07 by nakengelhardt