YosysHQ/yosys #2032: yosys master fails to build

FFY00, 2020-05-06:
## Steps to reproduce the issue

My nightly yosys arch package builder just failed.

Build instructions can be found here: https://github.com/FFY00/symbiflow-arch-pkgs/blob/master/yosys-git/PKGBUILD

## Actual behavior

```
/usr/bin/ld: src/base/abci/abc.o: in function `Abc_CommandTest':
/build/yosys-git/src/yosys/abc/src/base/abci/abc.c:13899: undefined reference to `Gia_EnumPerms'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:216: abc-default] Error 1
make: *** [Makefile:714: abc/abc-default] Error 2
```

Workflow run: https://github.com/FFY00/symbiflow-arch-pkgs/runs/648044973

FFY00, 2020-05-06:
Please let me know if this is a problem in ABC, I am building it in-tree.

Xiretza, 2020-05-06:
Yosys now uses a vendored fork of ABC: https://github.com/YosysHQ/yosys/blob/283b1130a651324ff870059dc3b1cf869948db93/Makefile#L141

The [yosys-git PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=yosys-git) on the AUR uses a different approach for the ABC submodule handling that caused a more obvious build failure (because the pinned ABCREV couldn't be checked out), and has since been updated.

FFY00, 2020-05-06:
I'll update the url. The sources should not be checked out in `prepare`.

Xiretza, 2020-05-06:
> The sources should not be checked out in `prepare`.

Off-topic here, but why? And where should it be done instead, `build()`?

FFY00, 2020-05-06:
They should be checked out by `makepkg` directly as specified in the `sources` array. The is a bit tricky for these cases. Checking them out in `prepare` goes against the guidelines but it isn't a particularly bad hack and I would say acceptable in this case. Although if it is possible to do it another way, that would be better.

Xiretza, 2020-05-06:
Oh, sure, that'd be best - however the [VCS Package Guidelines](https://wiki.archlinux.org/index.php/VCS_package_guidelines#Git_Submodules) actually recommend using `git submodule update` in `prepare()` for submodules, which is basically a checkout, so I wouldn't even be so sure about it being against guidelines.

FFY00, 2020-05-06:
Although somewhat antipattern, it is required in some cases.

whitequark, 2020-05-27:
@FFY00 Is this still an issue?

FFY00, 2020-06-26:
Sorry, I got lost on a few other things. This can be closed.

closed 2020-06-26 by FFY00