YosysHQ/yosys #2039: Verilog parameter propagation does not work

thomasdavidjansen, 2020-05-07:
Parameter propagation does not seem to be complete. The "OPERATION_WIDTH" parameter value from the "ALU_Test_Top" module seems to be correctly propagated to the line "assign o_carry = o_result[OPERATION_WIDTH - 1];" in the "ALU_Test" module but not to the "output wire [OPERATION_WIDTH - 1:0] o_result" line.

Setting the default value of the "OPERATION_WIDTH" parameter in the "ALU_Test" module to 32 (i.e. replacing "#(parameter OPERATION_WIDTH = 0)" with "#(parameter OPERATION_WIDTH = 32)") makes the warning go away.

The above indicates that the parameter value is not propagated to the line "output wire [OPERATION_WIDTH - 1:0] o_result" and the default value is used instead.

## Steps to reproduce the issue

Run the following command: yosys -f "verilog -sv" -p "synth_ice40 -top ALU_Test_Top -json ALU_Test_Top.json" ALU_Test.v ALU_Test_Top.v

with the below code:

**ALU_Test_Top.v**
```
`default_nettype none   // Don't allow undeclared nets
module ALU_Test_Top
(
    output wire [31:0] o_result,
    output wire o_carry
);
    ALU_Test #(.OPERATION_WIDTH(32)) alu //Using 32'd32 produces the same behavior
    (
        .i_operand0(32'd1),
        .i_operand1(32'd1),
        .o_result(o_result),
        .o_carry(o_carry)
    );
endmodule
```
**ALU_Test.v**
```
`default_nettype none   // Don't allow undeclared nets
module ALU_Test
#(parameter OPERATION_WIDTH = 0)
(
    input wire [OPERATION_WIDTH - 1:0] i_operand0,
    inout wire [OPERATION_WIDTH - 1:0] i_operand1,
    output wire [OPERATION_WIDTH - 1:0] o_result,
    output wire o_carry
);
    assign o_result = i_operand0 + i_operand1;
    assign o_carry = o_result[OPERATION_WIDTH - 1];
endmodule
```

## Expected behavior

No warnings

## Actual behavior

Getting a warning: "**Warning**: Range select out of bounds on signal `\o_result': Setting result bit to undef."

ZipCPU, 2020-05-08:
The warning is due to the fact that Yosys evaluates the submodule twice: first with its default parameters, for which it generates the warning, and then again a second time with the correct parameter values.

This is currently the way Yosys handles parameters.

eddiehung, 2020-05-08:
What @ZipCPU is saying is correct. Basically, when you do `./yosys -f "verilog -sv" file.v` you are implicitly doing a `read_verilog file.v`. What `read_verilog` does is it both analyses (i.e. builds the AST) and elaborates (i.e. converts into RTLIL) the design. Since there is no concept of `hierarchy` at this step, it elaborates each module in the design as if it was a top module, using default parameter values. It's this step that causes a warning.
```
$ ./yosys -q -f "verilog -sv" bug2039.v
bug2039.v:26: Warning: Range select out of bounds on signal `\o_result': Setting result bit to undef.
```

If you want to suppress this behaviour, you will need to add the `-defer` option and set the top explicitly:
```
$ ./yosys -q -f "verilog -sv -defer" bug2039.v -p "synth_ice40 -top ALU_Test_Top"
```

What this does is to defer the `elaboration` step to the `hierarchy` pass. Since nothing has been elaborated into RTLIL, `hierarchy` is currently unable to auto-detect the top-level (as it can't currently examine the AST) and so you have to set it manually.

thomasdavidjansen, 2020-05-08:
Thanks for the clarification. Although "-defer" solves the problem as expected, this is still a sort of usability issue. Could the warning maybe be extended to indicate that it might be that there is actually no issue with the Verilog code and that the warning cant be silenced with "-defer"?

Something like this: "Warning: Range select out of bounds on signal `\o_result': Setting result bit to undef. _Use the '-defer' flag to silence the warning if it is not an issue._"

closed 2020-05-08 by thomasdavidjansen

eddiehung, 2020-05-09:
I would say that the default parameters of a module should typically be valid for elaboration; there's no reason not to and using invalid parameters as a way of preventing parameter-less instantiation (or preventing use as a top module) is quite brittle.

Longer term, I would say that `-defer` should be enabled by default, but that would require `hierarchy` to be able to infer top modules by analysing the AST...

whitequark, 2020-05-09:
> that would require `hierarchy` to be able to infer top modules by analysing the AST...

Ideally not, since this would cause a problem in every case where the AST is not available, i.e. `connect_rpc`. But I think the frontend can emit an attribute that lists the *used modules* so that the `hierarchy` pass can do a topological sort without looking at AST or elaborating the modules.