YosysHQ/yosys #2043: pyosys API: Silent aborts (segfault/assert())

hackfin, 2020-05-11:
## Steps to reproduce the issue

A number of Python API calls can 'crash' the Python interpreter due to segfaults or assert() exits from the yosys kernel side.

I've collected them in a binder with a fairly recent pyosys build:

[image]

## Expected behavior

Return a Python exception.

## Actual behavior

Python interpreter or IPython kernel terminate without notice. When called from the shell, the error can be guessed. Inside IPython, the kernel dies non-verbosely and it seems impossible to debug the problem through stderr capture.

## Comment

Maybe it would be time to start a test suite for the pyosys API. Any ideas, opinions on `py.test`? Testing might get tricky due the module crashing (i.e. aborting the test)

hackfin, 2020-07-27:
Making another move on this one: Added docker hub py.test setup on this branch that is running a couple of basic tests.

https://github.com/hackfin/yosys/tree/pytest

Quickstart from local docker shell:
`docker run -it --rm  hackfin/yosys /home/pyosys/src/tests/pyosys/run-test.sh`

The logging test is currently disabled, to provoke the segfault, run

```
docker run -it --rm  hackfin/yosys bash -c "cd /home/pyosys/src/tests/pyosys && mv _test_logging.py test_logging.py && py.test-3"
```

However, it can be run as a single test without crash:

`docker run -it --rm  hackfin/yosys py.test-3  /home/pyosys/src/tests/pyosys/_test_logging.py`

As I don't really have the insights on the entire testing situation and status of the current Dockerfile, it's probably too early to do formal PRs. Also, I'm not sure about the future of dockerhub autobuilds.
On the travis side, it might likely just require an additional 'py.test' package installer line in the .yml to spool within `make test`

mmicko, 2026-08-24:
pyosys wrapper was redone using pybind11 but segfaults and asserts in kernel code are fatal errors so they could not be handled on python side

closed 2026-08-24 by mmicko