YosysHQ/yosys #2047: Create a RPC interface as an optional replacement to tcl

sequencer, 2020-05-13:
Currently, if user wanna interact with yosys, the only way is invoke a tcl shell to send command and receive command. The result is hard to parse, and string only without type struct informations.
So I wonder if possible to add a json/protobuf based RPC interface as a optional replacement to tcl which contains a more serializable API for other program to interact with yosys.

whitequark, 2020-05-13:
Have you seen pyosys?

sequencer, 2020-05-13:
Yes, I saw that, but I think that API is a python wrapper to tcl, which still doesn't have a standard API which embedded with Json/Protobuf or something else. Not sure if I'm right?

whitequark, 2020-05-13:
Nope. Unlike vendor toolchains Yosys isn't built around Tcl, and frankly the Tcl interface is an afterthought.

whitequark, 2020-06-04:
Closing since pyosys already implements this.

closed 2020-06-04 by whitequark