YosysHQ/yosys #2048: yosys-abc failed return code 127; libreadline6 vs libreadline7

gojimmypi, 2020-05-13:
I [commented in #1028](https://github.com/YosysHQ/yosys/issues/1028#issuecomment-627745567) on what I thought was a similar issue, but apparently it is unrelated, so I'm opening this new issue as [suggested](https://github.com/YosysHQ/yosys/issues/1028#issuecomment-627792593).

## Steps to reproduce the issue

I'm installing `yoysys` on a fresh WSL Ubuntu 18.04.4 LTS system per the [instructions](https://github.com/YosysHQ/yosys#setup).

## Expected behavior

`yoysys` to run without error.

## Actual behavior

As noted in https://github.com/icebreaker-fpga/icebreaker-examples/issues/10#issuecomment-627049625 I am seeing this error using the most recent version of `yosys` 0.9+2406 (git sha1 27b7ffc7, gcc 7.5.0-3ubuntu1~18.04 -fPIC -Os):

```
yosys -ql 7seg_count.log  -p 'synth_ice40 -top top -json 7seg_count.json' 7seg_count.v
ERROR: ABC: execution of command "/usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-zImuXR/abc.script 2>&1" failed: return code 127.
```

TL;DR: `yosys-abc` shows `libreadline6` is missing, but `libreadline7` is installed. How to:

- install `libreadline6`
or
- change dependency to `libreadline7`

`sudo apt-get install libreadline-dev` installs `libreadline7` but the most recent code in the master branch appears to be expecting `libreadline6`. _Older_ versions of `yosys` are expecting `libreadline7` and seem to work fine.

@daveshah1 [suggested](https://github.com/YosysHQ/yosys/issues/1028#issuecomment-627792593): `ldd /usr/local/bin/yosys-abc` and indeed that's pretty enlightening, indicating a missing `libreadline.so.6`:

```
  $ ldd /usr/local/bin/yosys-abc
        linux-vdso.so.1 (0x00007fffddd41000)
        libreadline.so.6 => not found
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fb37e3e0000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fb37e040000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fb37de30000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fb37dc20000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fb37d890000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fb37d660000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fb37d260000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fb37e800000)

$ yosys --version
Yosys 0.9+2406 (git sha1 27b7ffc7, gcc 7.5.0-3ubuntu1~18.04 -fPIC -Os)

$ apt list --installed | grep libread

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libreadline-dev/bionic,now 7.0-3 amd64 [installed]
libreadline7/bionic,now 7.0-3 amd64 [installed,automatic]

```

In contrast, I have an older `yoysys` working well on another WSL system: 93ef516d compiled in April, using `libreadline.so.7` (note the above newer NON-working system complains about missing `libreadline.so.6`). Here are the details if the older, working system:

```
$ yosys --version
Yosys 0.9+2406 (git sha1 93ef516d, clang 6.0.0-1ubuntu2 -fPIC -Os)

$ which yosys
/usr/local/bin/yosys

$ ls /usr/local/bin/yosys -al
-rwxr-xr-x 1 root root 18385608 Apr 12 11:44 /usr/local/bin/yosys

$ ldd /usr/local/bin/yosys-abc
        linux-vdso.so.1 (0x00007fffd922d000)
        libreadline.so.7 => /lib/x86_64-linux-gnu/libreadline.so.7 (0x00007f0491d20000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f0491b00000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f0491760000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0491550000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f0491340000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f0490fb0000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f0490d80000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f0490980000)
        libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007f0490750000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f0492000000)

$ apt list --installed | grep libread

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libreadline-dev/bionic,now 7.0-3 amd64 [installed]
libreadline5/bionic,now 5.2+dfsg-3build1 amd64 [installed]
libreadline7/bionic,now 7.0-3 amd64 [installed]
```

So the key seems to be the difference between non-working machine:
```
libreadline-dev/bionic,now 7.0-3 amd64 [installed]
libreadline7/bionic,now 7.0-3 amd64 [installed,automatic]
```
and the working machine:
```
libreadline-dev/bionic,now 7.0-3 amd64 [installed]
libreadline5/bionic,now 5.2+dfsg-3build1 amd64 [installed]
libreadline7/bionic,now 7.0-3 amd64 [installed]
```
How did I get `libreadline5/bionic,now 5.2+dfsg-3build1 amd64 [installed]`? 

As mentioned in https://github.com/YosysHQ/yosys/issues/643 I tried:

```
sudo apt-get purge libreadline-dev --assume-yes
sudo apt-get install libreadline6-dev --assume-yes
```
but no luck. I then end up with only:
```
apt list --installed | grep libread

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libreadline7/bionic,now 7.0-3 amd64 [installed,automatic]
```

Yes, it seems odd that  `sudo apt-get install libreadline6-dev` installs `libreadline7`, but here it is:

```
0 $ sudo apt-get purge libreadline-dev --assume-yes
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'libreadline-dev' is not installed, so not removed
The following package was automatically installed and is no longer required:
  libobjc-5-dev
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

0 $ sudo apt-get install libreadline6-dev --assume-yes
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libreadline-dev' instead of 'libreadline6-dev'
The following package was automatically installed and is no longer required:
  libobjc-5-dev
Use 'sudo apt autoremove' to remove it.
Suggested packages:
  readline-doc
The following NEW packages will be installed:
  libreadline-dev
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/133 kB of archives.
After this operation, 728 kB of additional disk space will be used.
Selecting previously unselected package libreadline-dev:amd64.
(Reading database ... 101287 files and directories currently installed.)
Preparing to unpack .../libreadline-dev_7.0-3_amd64.deb ...
Unpacking libreadline-dev:amd64 (7.0-3) ...
Setting up libreadline-dev:amd64 (7.0-3) ...
Processing triggers for install-info (6.5.0.dfsg.1-2) ...
```
and `yosys` still gives the same error.

If I copy the `/usr/local/bin/yosys*` and `/usr/local/share/yosys/` files from the working, April build machine to the non-working, current build machine... `yosys` works great. So the problem seems to be something changed in `yosys` since April.

The key to my broken `yosys` seems to be `libreadline.so.6` - but how to install this explicit version? Why doesn't the newer `libreadline7` meet the requirement? The [setup](https://github.com/YosysHQ/yosys#setup) indicates only `sudo apt-get install libreadline-dev`. But that, too installs `libreadline7`. 

I might have tried ` sudo apt-get purge libreadline7` but that was a bit frightening:

```
0 upgraded, 0 newly installed, 193 to remove and 0 not upgraded.
After this operation, 321 MB disk space will be freed.
``` 

Any help or suggestions will be appreciated. Thanks.

daveshah1, 2020-05-13:
Can you run `which make`, `which g++` and `echo $LDFLAGS` to rule out anything like breaking the environment (I've seen conda do stuff like this before).

gojimmypi, 2020-05-13:
```
0 $ which make
/usr/bin/make

0 $ make --version
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

0 $ which g++
/usr/bin/g++

127 $ g++ --version
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

0 $ echo $LDFLAGS

```
fwiw, I tried building `yosys` with each of these (separately, of course) with the same result for each:
```
$ make config-clang
$ make config-gcc
```

daveshah1, 2020-05-13:
Most mysterious, that all looks fine. Can you do a `ls -l abc/abc-*` inside the yosys tree to make sure that Yosys is actually rebuilding abc? And also post the output of `readlink /usr/lib/libreadline.so`.

gojimmypi, 2020-05-13:
that is mysterious. the file is days old, even after I do a `make clean`, it is still there:
```
$ ls -l abc/abc-*
-rwxrwxrwx 1 gojimmypi gojimmypi 52880469 May 10 10:47 abc/abc-fd2c9b1
```
and `readlink /usr/lib/libreadline.so` returns nothing (error code 1)

I kicked off another `make`... but it takes some time to complete....

daveshah1, 2020-05-13:
Ah, I'm not use to Debian style environments, it would be `readlink /usr/lib/x86_64-linux-gnu/libreadline.so` that you need to run.

gojimmypi, 2020-05-13:
```
 $   readlink /usr/lib/x86_64-linux-gnu/libreadline.so
/lib/x86_64-linux-gnu/libreadline.so.7
```
build is at 45%, two warnings:
```
[ 10%] Building frontends/verilog/verilog_parser.tab.cc
[ 11%] Building frontends/verilog/verilog_parser.tab.o
[ 11%] Building frontends/verilog/verilog_lexer.cc
[ 11%] Building frontends/verilog/verilog_lexer.o
frontends/verilog/verilog_lexer.cc: In function ‘int frontend_verilog_yylex(FRONTEND_VERILOG_YYSTYPE*, FRONTEND_VERILOG_YYLTYPE*)’:
frontends/verilog/verilog_lexer.cc:401:34: warning: ‘comment_caller’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 #define BEGIN (yy_start) = 1 + 2 *
                                  ^
frontends/verilog/verilog_lexer.l:132:6: note: ‘comment_caller’ was declared here

      ^
[ 11%] Building frontends/verilog/preproc.o
[ 12%] Building frontends/verilog/verilog_frontend.o
...
[ 40%] Building passes/techmap/libparse.o
[ 41%] Building passes/techmap/abc.o
passes/techmap/abc.cc: In member function ‘virtual void {anonymous}::AbcPass::execute(std::vector<std::__cxx11::basic_string<char> >, Yosys::RTLIL::Design*)’:
passes/techmap/abc.cc:1812:15: warning: ‘g_argidx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
      cmd_error(args, g_argidx, stringf("Unsupported gate type: %s", g.c_str()));
      ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 41%] Building passes/techmap/abc9.o
```

gojimmypi, 2020-05-13:
build complete.  `abc/abc-*` unrefreshed. Manually delete?
```
ls -l abc/abc-*
-rwxrwxrwx 1 gojimmypi gojimmypi 52880469 May 10 10:47 abc/abc-fd2c9b1
``` 

```
100%] Building techlibs/gowin/brams_init.mk
[100%] Building share/gowin/bram_init_16.vh
[100%] Building share/sf2/arith_map.v
[100%] Building share/sf2/cells_map.v
[100%] Building share/sf2/cells_sim.v

  Build successful.

gojimmypi@ElectronicsDesk : /mnt/c/workspace/yosys
0 $ sudo make install
[sudo] password for gojimmypi:
[Makefile.conf] CONFIG := gcc
mkdir -p /usr/local/bin
cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc /usr/local/bin
strip -S /usr/local/bin/yosys
strip /usr/local/bin/yosys-abc
strip /usr/local/bin/yosys-filterlib
mkdir -p /usr/local/share/yosys
cp -r share/. /usr/local/share/yosys/.
gojimmypi@ElectronicsDesk : /mnt/c/workspace/yosys
0 $ yosys --version
Yosys 0.9+2406 (git sha1 27b7ffc7, gcc 7.5.0-3ubuntu1~18.04 -fPIC -Os)

1 $ make test
[Makefile.conf] CONFIG := gcc
cd tests/simple && bash run-test.sh ""
make[1]: Entering directory '/mnt/c/workspace/yosys/tests/simple'
+ gcc -Wall -o /mnt/c/workspace/yosys/tests/tools/cmp_tbdata /mnt/c/workspace/yosys/tests/tools/cmp_tbdata.c
Test: arrays02 -> ok
Test: defvalue -> ERROR!
Note: Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog.
../tools/autotest.mk:10: recipe for target 'defvalue.sv' failed
make[1]: *** [defvalue.sv] Error 1
make[1]: Leaving directory '/mnt/c/workspace/yosys/tests/simple'
Makefile:754: recipe for target 'test' failed
make: *** [test] Error 2

```

gojimmypi, 2020-05-14:
@daveshah1  I have this resolved: I deleted the entire `yosys`, recloned it, and built it fresh.

I had done this before, but it may have been only in a prior Unbuntu version (I did a in-place upgrade to 18.04 when I could not get `yosys` working on older release).  Most recently I was only doing `make clean`.

I believe the root cause of this problem is `make clean` does not clean _everything_, which given your request that I look at `ls -l abc/abc-*` - perhaps you are already aware of this?

so the next experiment:

```
git clone https://github.com/YosysHQ/yosys.git yosys-clean
cd yosys-clean
rm .gitignore
make
make clean
git status
```
The only thing I saw remaining was a new `abc/` directory with a point-in-time clone of [abc](https://github.com/YosysHQ/abc)

I've created a new PR https://github.com/YosysHQ/yosys/pull/2049 with a proposal to delete all the `abc/` files at clean time.

Reminder that the `abc` checkout currently uses explicit version fd2c9b1 there.