YosysHQ/yosys #2060: Follow the XDG Base Directory Specification

piegamesde, 2020-05-15:
## Steps to reproduce the issue

Yosys should follow the [XDG Base Directory Specification](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html).

## Expected behavior

Please don't do that. Put it into `$XDG_DATA_HOME/yosys/` or `$XDG_CACHE_HOME/yosys/` instead.

## Actual behavior

I just found a `.yosys_history` file while cleaning up my home directory.

whitequark, 2020-05-15:
Yosys isn't an X11 or a desktop application.

Although in some cases (like Rust's [Cargo tool](https://github.com/rust-lang/cargo/issues/1734)) you can argue that the existing `~/.cargo` directory benefits from being split into configuration and cache directories so that existing tools don't e.g. unnecessarily backup the cache, Yosys isn't one of such cases: it only writes a history file.

piegamesde, 2020-05-15:
> Yosys isn't an X11 or a desktop application.

The XDG specifications grew out of their name a while ago and can be applied to all kinds of software, not only graphical/desktop/Xorg applications.

> it only writes a history file

That's great, because it will be easy to change. As software grows and gets older, more of those files  start appearing and migrating a big code base to the XDG specification afterwards is way harder than growing when appropriate code is already there. (See the Gradle folks, the `~/.gradle` is a few gigabytes of a mess and I doubt they'll ever make the jump.)

So yes, there may be less urge to do so for Yosys compared to other applications, but don't think it would hurt and there are still enough benefits to it to make it worth the effort.

whitequark, 2020-05-15:
What are the actual benefits in case of Yosys, besides "my `$HOME` is cleaner" which is entirely subjective, and "it'll cause more problems later" which is a slippery slope fallacy?

piegamesde, 2020-05-15:
Having all application data at only three locations makes it easier to manage. For example, it is more convenient to clean up files of removed software when it's not hidden between user data.

But yes, to me "my $HOME is cleaner" is the biggest point and all my arguments more or less boil down to this.

whitequark, 2020-05-20:
> The XDG specifications grew out of their name a while ago and can be applied to all kinds of software, not only graphical/desktop/Xorg applications.

I wasn't aware of the full breadth of software that uses XDG Base Directories these days, including `ps`, `nano`, `htop`, `hg` and so on. So I agree that with that context, it makes a lot of sense for Yosys to support them as well.

closed 2024-02-08 by mmicko