YosysHQ/yosys #2061: ice40: opt_lut can break carry packing by moving SB_CARRY out to another input than I3

smunaut, 2020-05-16:
## Steps to reproduce the issue

Here's a fairly minimal example that show the issue :

```
module bug #(
    parameter integer WIDTH = 12
)(  
    output reg  [WIDTH:0] cnt,
    input  wire clk,
    input  wire rst
);
    wire last_n;

    assign last_n = cnt[WIDTH];

    always @(posedge clk or posedge rst)
        if (rst)
            cnt <= 0;
        else
            cnt <= last_n ? ( cnt + { (WIDTH+1){last_n} } ) : 13'h1aaa;

endmodule
```

Build with : 

```
yosys -p 'synth_ice40  -json bug.json; show' bug.v
```

And if you look at the LUT that generates bit 12, you'll see the carry out of the previous bit is connected to I2 instead of I3.

This prevents nextpnr to properly pack the carry chain.

AFAICT this "bug" is added during the `opt_lut` pass. bit 12 doesn't have a SB_CARRY associated with it since we don't care about its carry output.

## Expected behavior

opt_lut should definitely optimize the two luts to a single one, but it should keep the carry out on I3.

## Actual behavior

This is the invalid conversion that's done :

[image]

smunaut, 2020-05-16:
From what I gather a similar issue can also happen at the LSB because the missing `CI` connection (in case there is not external carry in and it's forced to 1/0) would also prevent the `dlogic` system from `opt_lut` from triggering and protecting the `I1`/`I2` connections.

whitequark, 2020-05-16:
I think that maybe `dlogic` should be replaced with hardcoded logic for various families. IIRC @daveshah1 spoke out in favor of keeping it--what was the argument there?

daveshah1, 2020-05-16:
I don't remember that - in general I am in favour of arch-specific code over trying to make things totally generic (the same philosophy applies to nextpnr in general). There may have been a specific argument at the time that I've now forgotten, though.

whitequark, 2020-05-16:
OK, then I don't know anyone who is in favor of `-dlogic`. Let's replace that with arch-specific code.

smunaut, 2020-05-17:
Maybe I'm wrong, but I feel this would be way easier (and less of a guessing game) if the arithmetic mapping and the carry unwrap step was keeping a link like :  This is the LUT+CARRY that should be together and this is the carry from the previous/next bit ...

smunaut, 2021-07-10:
Just another even more extreme example of optimization messing up the carry chain. Here it manages break every single step of it, leading to twice as many LCs as it should ... ( for the final 12 bit adder/counter )

[bug.zip](https://github.com/YosysHQ/yosys/files/6795740/bug.zip)

```
Without workaround
Info: 	         ICESTORM_LC:    66/ 1280     5%

With workaround
Info: 	         ICESTORM_LC:    54/ 1280     4%
```

Obviously in this trivial example 12 more LCs isn't the end of the world, but
 - Still twice as many as that adder should take 
 - when I see this in real more full designs, they end up spread around, causing massive delays and kill f_max.
 - this is in a pretty classic counter with load pattern that I'd expect to see quite commonly used.

mwkmwkmwk, 2021-07-10:
*sigh*.

Timeline:

- 17th April 2019: change to lut mapping is applied that effectively maps the inputs in reverse (437fec0d88b4a2ad172edf0d1a861a38845f3b1d), breaking bit indices in `-dlogic`
- 16th July 2019: @smunaut submits a patch to reverse bit indices in `-dlogic` to match the above change (f28e38de9994151ea4e22608441dbc9e116d7b8c)
- 21st January 2020: change to lut mapping is reverted (ce6a690d27f3ce3b637f1d9be42b1efd744500d2), without reverting the `-dlogic` change

Resolution: revert f28e38de9994151ea4e22608441dbc9e116d7b8c.

smunaut, 2021-07-10:
Thanks. I'll test that ASAP.
I think it might fix that latest example, however I'm not 100% sure it will fix the original issue (where the bad optimization only happened on the first / last bit of the carry chain because it wasn't matching the strict pattern recognized by `dlogic` because no carry in/out for first/last in the chain).

mwkmwkmwk, 2021-07-10:
Alright, I pushed the change; if there are still cases with bad optimization, we'll reopen this bug

smunaut, 2021-07-10:
Yes, just tested and as expected, it fixes it for the "inner" bits.

However in the first example carry out still got moved from I3 to I2 in the last LUT preventing it being packed properly. 

The second example is still not the same :

```
Without workaround
Info: 	         ICESTORM_LC:    57/ 1280     4%

With workaround
Info: 	         ICESTORM_LC:    55/ 1280     4%
```

However it's a bit less clear why and I don't think opt_lut is at fault.
The first carry isn't packed with the first LUT but that might be a `nextpnr` issue because AFAICT, it could be from the netlist.
And then there is one more LUT4 used somewhere ...

So still an issue there and the test case in the first comment of the issue is still applicable.

commit 726fabd6 (2021-07-10): ice40: Fix LUT input indices in opt_lut -dlogic (again).

Fixes #2061.
PR #2897 (2021-07-29): opt_lut: Allow more than one -dlogic per cell type.
Fixes #2061.
commit 54e75129 (2021-07-29): opt_lut: Allow more than one -dlogic per cell type.

Fixes #2061.
closed 2021-07-29 by mwkmwkmwk