YosysHQ/yosys #2065: ECP5 and Xilinx synthesis hits issue in kernel/rtlil.cc:867

antonblanchard, 2020-05-18:
We are hitting an issue in yosys when building the rotator in microwatt. To make it easier to reproduce, I've converted the VHDL to Verilog by passing it through yosys.

To reproduce:

```
yosys -p "read_verilog rotator.v; synth_ecp5"
```

Which hits:

```
ERROR: Found error in internal cell \rotator.$auto$alumacc.cc:485:replace_alu$1048.A_conv ($pos) at kernel/rtlil.cc:867:
  attribute \module_not_derived 1
  attribute \src "/home/anton_unencrypted/yosys/share/ecp5/arith_map.v:39.69-39.94"
  cell $pos $auto$alumacc.cc:485:replace_alu$1048.A_conv
    parameter \Y_WIDTH 5
    parameter \A_WIDTH 0
    parameter \A_SIGNED 0
    connect \Y $auto$alumacc.cc:485:replace_alu$1048.A_buf
    connect \A $auto$alumacc.cc:485:replace_alu$1048.A
  end
```

Both ecp5 and xilinx hit this.

antonblanchard, 2020-05-18:
```
module rotator(rs, ra, shift, insn, is_32bit, right_shift, arith, clear_left, clear_right, result, carry_out);
  wire _000_;
  wire _001_;
  wire _002_;
  wire _003_;
  wire _004_;
  wire _005_;
  wire _006_;
  wire _007_;
  wire _008_;
  wire _009_;
  wire _010_;
  wire _011_;
  wire _012_;
  wire _013_;
  wire _014_;
  wire _015_;
  wire _016_;
  wire _017_;
  wire _018_;
  wire _019_;
  wire _020_;
  wire _021_;
  wire _022_;
  wire _023_;
  wire _024_;
  wire _025_;
  wire _026_;
  wire _027_;
  wire _028_;
  wire _029_;
  wire _030_;
  wire _031_;
  wire _032_;
  wire _033_;
  wire _034_;
  wire _035_;
  wire _036_;
  wire _037_;
  wire _038_;
  wire _039_;
  wire _040_;
  wire _041_;
  wire _042_;
  wire _043_;
  wire _044_;
  wire _045_;
  wire _046_;
  wire _047_;
  wire _048_;
  wire _049_;
  wire _050_;
  wire _051_;
  wire _052_;
  wire _053_;
  wire _054_;
  wire _055_;
  wire _056_;
  wire _057_;
  wire _058_;
  wire _059_;
  wire _060_;
  wire _061_;
  wire _062_;
  wire _063_;
  wire _064_;
  wire _065_;
  wire _066_;
  wire _067_;
[... 622 lines cut ...]
  assign _156_ = _155_ ? 1'h1 : 1'h0;
  assign _157_ = $signed(32'd43) <= $signed({ 25'h0000000, me });
  assign _158_ = _157_ ? 1'h1 : 1'h0;
  assign _159_ = $signed(32'd44) <= $signed({ 25'h0000000, me });
  assign _160_ = _159_ ? 1'h1 : 1'h0;
  assign _161_ = $signed(32'd45) <= $signed({ 25'h0000000, me });
  assign _162_ = _161_ ? 1'h1 : 1'h0;
  assign _152_ = ~ clear_left;
  assign _164_ = $signed(32'd46) <= $signed({ 25'h0000000, me });
  assign _165_ = _164_ ? 1'h1 : 1'h0;
  assign _166_ = $signed(32'd47) <= $signed({ 25'h0000000, me });
  assign _163_ = clear_right & _152_;
  assign _167_ = _166_ ? 1'h1 : 1'h0;
  assign _168_ = $signed(32'd48) <= $signed({ 25'h0000000, me });
  assign _169_ = _168_ ? 1'h1 : 1'h0;
  assign _170_ = $signed(32'd49) <= $signed({ 25'h0000000, me });
  assign _171_ = _170_ ? 1'h1 : 1'h0;
  assign _172_ = $signed(32'd50) <= $signed({ 25'h0000000, me });
  assign _173_ = _172_ ? 1'h1 : 1'h0;
  assign _175_ = $signed(32'd51) <= $signed({ 25'h0000000, me });
  assign _176_ = _175_ ? 1'h1 : 1'h0;
  assign result = _230_;
  assign carry_out = _235_;
endmodule
```

whitequark, 2020-05-18:
cc @mwkmwkmwk

daveshah1, 2020-05-18:
See also #2058. If this is going to take a while to be fixed, maybe we should revert #2027 so master remains usable?

antonblanchard, 2020-05-18:
I can confirm that reverting e936ac61ea22 "ast: swap range regardless of range_left >= 0" fixes it.

mwkmwkmwk, 2020-05-18:
This is a duplicate of #2058, yes.

mwkmwkmwk, 2020-05-18:
Should be fixed by #2067, can you try that out?

antonblanchard, 2020-05-19:
Thanks @mwkmwkmwk, that does fix my issue.

closed 2020-05-19 by mwkmwkmwk