YosysHQ/yosys #2071: ERROR: Assert `no_loops' failed in

kost, 2020-05-22:
## Steps to reproduce the issue

Standard build of SNES_MiSTer_ulx3s
```
git clone https://github.com/daveshah1/SNES_MiSTer_ulx3s
cd SNES_MiSTer_ulx3s
cd ulx3s/scripts
bash build.sh
```

outputs:

```
Warning: found logic loop in module top:
    cell main.SNES.smp.spc700.alu.35409 ($pmux)
    wire \main.SNES.smp.spc700.alu.addsub.b [7]
Warning: logic loop in mux tree at signal \main.SNES.cpu.p65_a [23:4] in module top.
Warning: logic loop in mux tree at signal \main.SNES.cpu.p65c816.addrgen.newaah in module top.
Warning: logic loop in mux tree at signal \main.SNES.cpu.p65c816.addrgen.newaal in module top.
Warning: logic loop in mux tree at signal \main.SNES.cpu.p65c816.alu.intr8 in module top.
Warning: logic loop in mux tree at signal \main.SNES.cpu.p65c816.mcode.state in module top.
Warning: logic loop in mux tree at signal \main.SNES.cpu.p65c816.alu.result8 in module top.
Warning: logic loop in mux tree at signal \main.SNES.smp.spc700.addrgen.newal in module top.
ERROR: Assert `no_loops' failed in passes/techmap/abc9_ops.cc:754.
```

Yosys version (latest git):
```
# yosys -V
Yosys 0.9+2406 (git sha1 4f0f3211, g++ 8.3.0 -Os)
```

## Expected behavior

Expected behavior is to build without errors. it builds correctly without -abc9 

## Actual behavior

```
ERROR: Assert `no_loops' failed in passes/techmap/abc9_ops.cc:754.
```

eddiehung, 2020-05-22:
It looks like this needs the GHDL plugin. Can you provide the testcase as a Verilog or ILANG file please?

kost, 2020-05-23:
I'm not sure I can, but maybe @daveshah1 can help since it's his baby.

daveshah1, 2020-05-23:
Please see ILANG attached, run `yosys -p "synth_ecp5 -top top -abc9" snes.il`

[snes.zip](https://github.com/YosysHQ/yosys/files/4671436/snes.zip)

eddiehung, 2020-05-25:
So it looks like there are some combinatorial loops inside this design:
```
2.10. Executing CHECK pass (checking for obvious problems).
checking module top..
Warning: found logic loop in module top:
    cell main.SNES.cpu.p65c816.22025 ($pmux)
    wire \main.SNES.cpu.p65c816.mcode.state [0]
Warning: found logic loop in module top:
    cell main.SNES.cpu.p65c816.22025 ($pmux)
    wire \main.SNES.cpu.p65c816.mcode.state [1]
Warning: found logic loop in module top:
    cell main.SNES.cpu.p65c816.22025 ($pmux)
    wire \main.SNES.cpu.p65c816.mcode.state [2]
Warning: found logic loop in module top:
    cell main.SNES.cpu.p65c816.22025 ($pmux)
    wire \main.SNES.cpu.p65c816.mcode.state [3]
Warning: found logic loop in module top:
    cell main.SNES.cpu.p65c816.22995 ($pmux)
    wire \main.SNES.cpu.p65_a [0]
Warning: found logic loop in module top:
    cell main.SNES.cpu.p65c816.22995 ($pmux)
    wire \main.SNES.cpu.p65_a [1]
<<SNIP>>
```
Nonetheless `abc9` should be handling it more gracefully; should be fixed in #2082.

daveshah1, 2020-05-26:
These combinational loops didn't exist before, they might be related to the fairly new $pmux support in ghdl - or they might be a false path.

PR #2082 (2020-06-04): abc9: fixes around handling combinatorial loops
Fixes #2071.
closed 2020-06-04 by eddiehung