YosysHQ/yosys #2114: systemverilog typedef enum in package can not be used in interface

zinsserzh, 2020-06-06:
## Steps to reproduce the issue

```systemverilog
// top.sv

package pkg;
    typedef enum bit [1:0] {
        CODE_NOOP = 2'd0,
        CODE_PASS = 2'd1,
        CODE_FAIL = 2'd2,
        CODE_DONE = 2'd3
    } status_code;
endpackage

module top(
    input [1:0] sin,
    output [1:0] sout,
);
    my_if u_my_if();
    a u_a(.sin, .my_if(u_my_if));
    b u_b(.sout, .my_if(u_my_if));
endmodule

interface my_if();
    pkg::status_code status;
    modport in  (input  status);
    modport out (output status);
endinterface

module a(
    input [1:0] sin,
    my_if.out my_if
);
    assign my_if.status = sin;
endmodule

module b(
    output [1:0] sout,
    my_if.in my_if
);
    assign sout = my_if.status;
endmodule
```

## Expected behavior

If `pkg::status_code status;` is changed to `bit [1:0] status;` verilog frontend is working fine
```
 Yosys 0.9+2406 (git sha1 bd2ecc2d, gcc 7.5.0-3ubuntu1~18.04 -fPIC -Os)

yosys> read_verilog -sv top.sv
1. Executing Verilog-2005 frontend: top.sv
Parsing SystemVerilog input from `top.sv' to AST representation.
Generating RTLIL representation for module `\top'.
Generating RTLIL representation for module `\my_if'.
Generating RTLIL representation for module `\a'.
top.sv:30: Warning: Identifier `\my_if.status' is implicitly declared.
Generating RTLIL representation for module `\b'.
top.sv:37: Warning: Identifier `\my_if.status' is implicitly declared.
Successfully finished Verilog frontend.
```

## Actual behavior

From yosys master
```
 Yosys 0.9+2406 (git sha1 bd2ecc2d, gcc 7.5.0-3ubuntu1~18.04 -fPIC -Os)

yosys> read_verilog -sv top.sv
1. Executing Verilog-2005 frontend: top.sv
Parsing SystemVerilog input from `top.sv' to AST representation.
Generating RTLIL representation for module `\top'.
Generating RTLIL representation for module `\my_if'.
top.sv:22: ERROR: Unknown identifier `\pkg::status_code' used as type name
```

dh73, 2020-06-08:
Just a note for the provided example.
For enumerated types, IEEE 1800-2012 in the `6.19.3 Type Checking` section:

> Enumerated types are strongly typed; thus, a variable of type enum cannot be directly assigned a value that lies outside the enumeration set unless an explicit cast is used or unless the enum variable is a member of a union.

By this specific lines:
```verilog
input /*wire*/ [1:0] sin,
assign my_if.status = sin;
```
You're assigning a non enumerated type variable to an enum, violating the SystemVerilog strongly typed rule. The synthesis tool can do one of two things in this case:
1) Automatically cast the `sin` variable.
2) Blackboxing the  `a` module due violation of this rule.

Obviously, none of those behaviors are safe to ignore.

To use enumerated types as top ports, if none of the drivers are enum types, you'll need to cast them, and that's not yet supported in Yosys.

zachjs, 2020-08-01:
@zinsserzh You may want to try https://github.com/zachjs/sv2v, which can convert the given example into Verilog supported by Yosys.

ESP2866, 2022-04-14:
I have a similar issue but with  typedef structs:

package WBpack;

typedef struct packed{
	logic [3:0] adX; // Register number of first register to be written back
	logic [3:0] adY; // Register number of second register to be written back
	logic weX; // Should register X be written
	logic weY; // Should register Y be written
} CtrlWB;

endpackage : WBpack

package MEMpack;

typedef struct packed{
	logic [31:0] resultX;
	logic [31:0] resultY;
	WBpack::CtrlWB ctrlWB;
} MEMout;
endpackage : MEMpack

where the error is ERROR: typedef for user type `\WBpack::CtrlWB' not found

Version: Yosys 0.16+31 (git sha1 c1646a00a, x86_64-w64-mingw32-g++ 9.2.1 -Os)

daglem, 2023-01-23:
> I have a similar issue but with typedef structs:

This should be solved by #3640.

PR #3658 (2023-02-12): Resolve package types in interfaces
Fixes #2114
closed 2023-02-12 by zachjs