YosysHQ/yosys #2127: cxxrtl RFE: helper methods to cast internal data representation to bool, integer, etc.

tomverbeure, 2020-06-09:
Right now, when the main wrapper around a CXXRTL model wants to access the value of a port or signal, it needs to dive into the internal data representation of the model.

Like this:
```c
        int cur_led = top.p_led.curr.data[0];
```

For a cxxrtl user, `curr` and `data` are implementation details that should be hidden and that might change as internal specializations are implemented (e.g to increase performance)

CXXRTL needs some casting function remove the need to do this:

* convert single-bit wires to bool
* convert vectors up to 64-bit to int64_t or uint64_t.
* methods to access values in 64-bit chunks for vectors larger than 64 bits
* methods to access locations inside a memory

whitequark, 2020-06-09:
For context, the API you are currently using is a public one. It is a fundamental part of CXXRTL's design and it is extremely unlikely that it will ever be changed, including the `uint32_t` chunk size. There are a few auxiliary structures that I've described as "internal", but `value`, `wire`, `memory`, `metadata` and `module` are all public.

So I consider this issue a "quality of life" enhancement rather than "API stability" enhancement.

> * convert single-bit wires to bool

:+1:

> * convert vectors up to 64-bit to int64_t or uint64_t.

:+1:

> * methods to access values in 64-bit chunks for vectors larger than 64 bits

How would this work? `value.data` is not currently 32-bit-aligned, so I can't return a pointer to the chunks as `uint64_t*` (plus that would violate strict aliasing). If you just need to access values in *some* chunks, you can already do that in 32-bit chunks through the `.data` array.

> * methods to access locations inside a memory

Already exists: `mem[index]` for reads, `mem[index] = value<N>{...}` for writes **ONLY to ROMs before the first delta cycle**, `mem.update(index, val, mask, prio);` for writes without race conditions during the simulation.

tomverbeure, 2020-06-09:
> How would this work? value.data is not currently 64-bit-aligned, so I can't return a pointer to the chunks as uint64_t* (plus that would violate strict aliasing). If you just need to access values in some chunks, you can already do that in 32-bit chunks through the .data array.

If the current data structure is considered frozen, then there's indeed not much to improve on that front.

I suspect that most people will want to know the current value, not the next one, so a way to avoid having to include `curr` might be useful.

whitequark, 2020-06-09:
> If the current data structure is considered frozen, then there's indeed not much to improve on that front.

It's unlikely to change, but not because it's frozen or because of ABI stability. It's because I benchmarked `int32_t` against `intptr_t` and found no runtime benefit to the latter, whereas the space overhead is easy to see. If you have good evidence in favor of `intptr_t` we can switch to it.

> I suspect that most people will want to know the current value, not the next one, so a way to avoid having to include `curr` might be useful.

Right, I'm thinking about adding accessors like `.get<int>()` and `.set<int64_t>()` to both `value` (where they would return and change the value itself) and `wire` (where they would return from `curr` and change `next`, correspondingly). Does that sound good to you?

whitequark, 2020-06-09:
> If you have good evidence in favor of `intptr_t` we can switch to it.

(There's also the fact that having chunk size different depending on the architecture carries the inherent hazard that people would accidentally depend on a specific chunk size. So the performance benefit here must be large enough to outweigh that drawback, not just nonzero.)

tomverbeure, 2020-06-09:
32-bit is probably much better from a memory usage point of view.

whitequark, 2020-06-09:
Right, that's true, what I mean is that I also found no improvement in runtime. Of course my benchmarking could be flawed so if that turns out to be the case and the current tradeoff turns out to be bad I'm open to changing it.

tomverbeure, 2020-06-09:
Ok! Let's just stay at 32 bit.

whitequark, 2020-06-18:
Please try using https://github.com/YosysHQ/yosys/pull/2171 and report results.

PR #2171 (2020-06-19): cxxrtl: add .get() and .set() accessors on value<> and wire<>
For several reasons:
  * They're more convenient than accessing .data.
  * They accommodate variably-sized types like `size_t` transparently.
  * They statically ensure that no out of range conversions happen.

For now these are only provided for unsigned integers, but eventually they should be provided for signed integers too. (Annoyingly this affects conversions to/from `char` at the moment.)

Fixes #2127.
commit 962a2f3b (2020-06-19): cxxrtl: add .get() and .set() accessors on value<> and wire<>.

For several reasons:
  * They're more convenient than accessing .data.
  * They accommodate variably-sized types like size_t transparently.
  * They statically ensure that no out of range conversions happen.

For now these are only provided for unsigned integers, but eventually
they should be provided for signed integers too. (Annoyingly this
affects conversions to/from `char` at the moment.)

Fixes #2127.
closed 2020-06-19 by whitequark