YosysHQ/yosys #2129: Assertion failure in kernel/rtlil.cc when attempting to write_cxxrtl

dan-rodrigues, 2020-06-09:
As requested:

## Steps to reproduce the issue

`yosys -p 'read_ilang ics32_ilang; write_cxxrtl -O0 sim.cpp'`

Using the following as input:
[ics32_ilang.zip](https://github.com/YosysHQ/yosys/files/4751382/ics32_ilang.zip)

## Expected behavior

Outputs sim.cpp without error

## Actual behavior

```
ERROR: Assert `is_fully_const() && GetSize(chunks_) <= 1' failed in kernel/rtlil.cc:3801.`
```

## Version

Yosys built from c172b30c24285faa2b57f84a2a5162c5f3dcfb4c

whitequark, 2020-06-09:
This has no bearing on this issue (it is clearly a bug) but please note that using levels lower than `-O2` with cxxrtl has virtually no benefits and it will cause the generated C++ to stress your C++ compiler a lot.

whitequark, 2020-06-09:
The direct cause of the issue is that some `$meminit` cells did not have constant address/data due to the specific script used. Of course this shouldn't assert. But I'm not sure where the code to check for that should live; right now it's in `memory_collect`, but cxxrtl specifically does *not* want `$mem` cells. @clairexen opinions?

whitequark, 2020-11-01:
Will be fixed by #2425 and print:

```
ERROR: Non-constant address $memwr$\alpha_lut$vdp/vdp_blender.v:70$2783_ADDR in memory initialization $meminit$\alpha_lut$vdp/vdp_blender.v:0$3068.
```

whitequark, 2020-11-01:
(The message isn't super great but any fix should be in the common memory code so that non-cxxrtl users benefit from it as well.)

mwkmwkmwk, 2020-11-01:
Also FYI: the solution here (assuming the input is valid in the first place) is to run `opt_expr` to perform missing constant folding before running the backend.

PR #2425 (2020-11-01): cxxrtl: don't assert on non-constant $meminit inputs
Fixes #2129.
commit 2ba05f5c (2020-11-01): cxxrtl: don't assert on non-constant $meminit inputs.

Fixes #2129.
closed 2020-11-01 by whitequark