YosysHQ/yosys #2157: cxxrtl: signed shr doesn't sign-extend correctly for values smaller than chunk size

miek, 2020-06-14:
When (signed) shr handles the final chunk
https://github.com/YosysHQ/yosys/blob/74e93e083ff23f3381fe2409e5847f9843840b17/backends/cxxrtl/cxxrtl.h#L382

if, for example, chunk::bits is 32, Bits is 16, and the shift is 1 then this will only fill bit 31 of the chunk.
The RHS should be equivalent to `chunk::mask << (Bits - shift_bits)` when the value is smaller than the chunk size.

Also I think there's an off-by-one, it should assign to `result.data[chunks - shift_chunks - 1]`.

whitequark, 2020-06-14:
Yep, that line is a bug--I actually noticed it this morning. Care to send a PR?

miek, 2020-06-14:
Yep, will do. I need to think about it some more now though, I just realised this applies to larger values too. I think it hits any value that's not a multiple of the chunk size, when the shift is less than the chunk size.

PR #2158 (2020-06-15): cxxrtl: fix sshr sign-extension.
This fixes #2157

I think this covers all the cases now. It works by finding the most significant chunk after the shift, calculating the number of bits in that chunk, then masking the rest. In the edge case where a small shift lands right on a chunk boundary, it will mask off the whole chunk above - I think that's reasonable so it didn't seem worth adding a special case.

This won't guarantee masking all chunks above the new "valid" region (ie: if a small shift moves the value out of chunk[1] and completely into chunk[0], this won't mask chunk[1]). I wasn't sur [... cut]
closed 2020-06-15 by whitequark