YosysHQ/yosys #2161: ICE40 ABC issue?

splhack, 2020-06-15:
## Steps to reproduce the issue

I'm trying to output VGA signal on TinyFPGA BX (ICE40LP8K).
The signal doesn't work only with default `ABC`, probably due to some timing, even though it worked before.

### Results on master 74e93e083ff23f3381fe2409e5847f9843840b17
synth_ice40  option|Result
------------ | -------------
(Default = ABC) | :x:
`-retime`| :white_check_mark:
`-noabc`| :white_check_mark:
`-flowmap`| :white_check_mark:
`-abc2`| :white_check_mark:
`-abc9`| :white_check_mark:

But if I revert this commit  https://github.com/YosysHQ/yosys/commit/1679682fa3ae18282b49452891282901a3548ecc like this (identified by git bisect), VGA signal works even with `ABC`.

    diff --git a/passes/opt/opt_reduce.cc b/passes/opt/opt_reduce.cc
    index f640f50a..e591a30e 100644
    --- a/passes/opt/opt_reduce.cc
    +++ b/passes/opt/opt_reduce.cc
    @@ -44,10 +44,9 @@ struct OptReduceWorker
                    cells.erase(cell);

                    RTLIL::SigSpec sig_a = assign_map(cell->getPort(ID::A));
    -               sig_a.sort_and_unify();
                    pool<RTLIL::SigBit> new_sig_a_bits;

    -               for (auto &bit : sig_a)
    +               for (auto &bit : sig_a.to_sigbit_set())
                    {
                            if (bit == RTLIL::State::S0) {
                                    if (cell->type == ID($reduce_and)) {
    @@ -87,7 +86,6 @@ struct OptReduceWorker
                    }

                    RTLIL::SigSpec new_sig_a(new_sig_a_bits);
    -               new_sig_a.sort_and_unify();

                    if (new_sig_a != sig_a || sig_a.size() != cell->getPort(ID::A).size()) {
                            log("    New input vector for %s cell %s: %s\n", cell->type.c_str(), cell->name.c_str(), log_signal(new_sig_a));

### Design
- [top.zip](https://github.com/YosysHQ/yosys/files/4778615/top.zip)
  - The verilog top.v was generated by https://github.com/phanrahan/magma
  - Just in case, the zip has working output (good.json), and non-working output (bad.json)
  - (Expected result is screen filled with red by VGA output signal)

### Versions
- Yosys 0.9+2406 (git sha1 74e93e08, clang 10.0.0-4ubuntu1 -fPIC -Os)
- nextpnr-ice40 -- Next Generation Place and Route (Version c9e7d14)
- icepack cd2610e0fa1c6a90e8e4e4cfe06db1b474e752bb

### Build
    yosys -q -p 'synth_ice40 -top top -json top.json' top.v
    nextpnr-ice40 --lp8k --package cm81 --json top.json --pcf TinyFPGA-BX.pcf --asc top.asc --opt-timing
    icepack top.asc top.bin

ZipCPU, 2020-06-15:
When you say the signal "doesn't work", can you be more specific?

I've chased several bugs down through ABC, and they've most often been associated with registers whose timing wasn't explicitly driven--things like not using SB_IO's for outputs where timing was important.  The yosys change just revealed the bug in the underlying user design, rather than actually being broken itself.

Dan

eddiehung, 2020-06-15:
See https://github.com/YosysHQ/yosys/pull/1682#issuecomment-583097452

TL;DR: Like simulated annealing placement, ABC's default techmapper is chaotic in that it can return different solutions -- some better, some worse -- depending on the logical function and/or its input ordering (the latter being akin to a placer's random seed). Unfortunately, even ABC's author through the comment above accepts this as a known limitation.
One thing I have thought about is to expose a "seed" option in Yosys that re-orders these inputs such that at least you can sweep this solution space.

splhack, 2020-06-16:
thank you for the clarification!

closed 2020-06-16 by splhack