YosysHQ/yosys #2166: cxxrtl backend generates non-working c++ sources when NDEBUG=1

madebr, 2020-06-17:
## Steps to reproduce the issue

1. Build yosys using:
```
make 'ENABLE_NDEBUG=1'
```
2. create design **blinky.v**
```verilog
`default_nettype	none
module blinky(i_clk, o_led);
`ifdef	SIMULATION
	parameter	WIDTH = 12;
`else
	parameter	WIDTH = 27;
`endif
	input	wire	i_clk;
	output	wire	o_led;
	reg	[WIDTH-1:0]	counter;
	always @(posedge i_clk)
		counter <= counter + 1'b1;
	assign	o_led = counter[WIDTH-1];
endmodule
```
3. Create script **proc.ys**
```
verilog_defines -DSIMULATION
read_verilog ./blinky.v
hierarchy -check -top blinky
write_ilang BlinkyTop.sim.ilang
write_cxxrtl -header BlinkyTop.sim.cpp
```
4. Execute
```
yosys proc.ys
```

Then the generated sources are non-functional.

## Expected behavior

*Working* c++ sources

## Actual behavior

<details><summary>BlinkyTop.sim.h</summary>
<p>

```c++
#ifndef CXXRTL_DESIGN_HEADER
#define CXXRTL_DESIGN_HEADER

#include <backends/cxxrtl/cxxrtl_capi.h>

#ifdef __cplusplus
extern "C" {
#endif

cxxrtl_toplevel cxxrtl_design_create();

#ifdef __cplusplus
}
#endif

#ifdef __cplusplus

#include <backends/cxxrtl/cxxrtl.h>

using namespace cxxrtl;

namespace cxxrtl_design {

} // namespace cxxrtl_design

#endif // __cplusplus

#endif
```

</p>
</details>

<details><summary>BlinkyTop.sim.cpp</summary>
<p>

```c++
#include "BlinkyTop.sim.h"

#if defined(CXXRTL_INCLUDE_CAPI_IMPL) || \
    defined(CXXRTL_INCLUDE_VCD_CAPI_IMPL)
#include <backends/cxxrtl/cxxrtl_capi.cc>
#endif

#if defined(CXXRTL_INCLUDE_VCD_CAPI_IMPL)
#include <backends/cxxrtl/cxxrtl_vcd_capi.cc>
#endif

using namespace cxxrtl_yosys;

namespace cxxrtl_design {

} // namespace cxxrtl_design

cxxrtl_toplevel cxxrtl_design_create() {
	return new _cxxrtl_toplevel { std::unique_ptr<cxxrtl_design::p_blinky>(new cxxrtl_design::p_blinky) };
}
```

</p>
</details>

whitequark, 2020-06-17:
I can reproduce this, thanks for the report.

whitequark, 2020-06-17:
See https://github.com/YosysHQ/yosys/pull/2167.

PR #2167 (2020-06-18): cxxrtl: don't compute vital values in log_assert()
This breaks NDEBUG builds.

Fixes #2166.
commit 3c4e974d (2020-06-23): cxxrtl: don't compute vital values in log_assert().

This breaks NDEBUG builds.

Fixes #2166.
closed 2020-06-18 by whitequark