YosysHQ/yosys #2176: Attributes get lost when passing through ABC9

acomodi, 2020-06-19:
## Steps to reproduce the issue

[test_loc.tar.gz](https://github.com/YosysHQ/yosys/files/4804986/test_loc.tar.gz)
Yosys version https://github.com/YosysHQ/yosys/commit/ede4b10da8fdbdcff900b93c6c723516901483ff.

1. extract the tar
2. run `./run.sh`: this should return `LOC constraint is present!` at the end of the execution
3. run `./run_abc9.sh`: this should return `LOC constraint is not present!` at the end of the execution.

## Expected behavior

LOC attribute expressed in the design (e.g. `(* LOC="<location>" *)`) should not get trimmed when running through ABC9.

## Actual behavior

LOC attribute get trimmed when passing through ABC. This should be related to the fact that FF do get passed through ABC9, causing a possible loss of attributes during map/unmap operations.

ZirconiumX, 2020-06-22:
cc @eddiehung

eddiehung, 2020-06-23:
I'll try and look at it on the weekend.

eddiehung, 2020-07-08:
Testcase is not a valid gzip file.

eddiehung, 2020-07-08:
```
read_verilog <<EOF
module top(input clk, D, output Q);
(* LOC="FOO" *)
FDRE ff(.C(clk), .D(D), .Q(Q));
endmodule
EOF
synth_xilinx -noiopad -noclkbuf -abc9 -dff
dump top
```
works fine for me.

acomodi, 2020-07-09:
@eddiehung Yeah, I forgot to close this issue actually. This issue got solved as I do not get the error anymore as well. I believe this was somehow related to https://github.com/YosysHQ/yosys/issues/2204. Closing

closed 2020-07-09 by acomodi