YosysHQ/yosys #2183: ERROR: Mismatch in directionality for cell port...

davidthings, 2020-06-22:
## Steps to reproduce the issue

Possibly the same issue as https://github.com/YosysHQ/yosys/issues/2053

My code uses arrays where some lines are inputs and some are outputs.  With recent Yosys builds, some small part of this code is that used to work for a long time is no-longer compiling.   

Boiled down, the following snippet illustrates this.

There is a 9-bit array, which is marked as `inout` because of the mix of signals.  The subordinate module, `worker` assigns the parts of the bundle.

``` verilog

module worker(
        inout  [8:0] data_in,
        inout  [8:0] data_out
    );
    assign data_out[7:0] = data_in[7:0];
    assign data_in[ 8 ] = data_out[ 8 ];
endmodule

module tests_tbx (
        inout [8:0] data
    );
    wire [8:0] data_shadow;
    assign data_shadow[7:0] = 8'H55;
    worker w( data_shadow, data );
endmodule

```
## Expected behavior

This should compile.

## Actual behavior

```
$ make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port tests_tbx.w.data_in: \data_shadow <= \w.data_in
Makefile:30: recipe for target 'tests_tbx.json' failed
make: *** [tests_tbx.json] Error 1
```

whitequark, 2020-06-22:
Do you think you could bisect the issue? I don't have any hardware performant enough to make that reasonably fast so it'll be a lot easier for me to look into it if someone else narrowed down the problem.

davidthings, 2020-06-22:
Damn!  I was feeling so proud that I got it down to that small from my 10k code base!  

Can you say a little more about how I could help further?

whitequark, 2020-06-22:
Having a minimal test case is super helpful! I suspect (but don't know for sure) that the bug is introduced somewhere between f57524cf713e2cdc368ded8b7e13f85d310b8749 and 8c4cb1885b2cb6b7b26d7b9b7113e174c0eefffd (current master). Could you bisect Yosys to narrow it down to a single commit?

davidthings, 2020-06-22:
Oh!  That.  Yes.  I'll give it a shot.  (I thought you meant bisect the example!)

davidthings, 2020-06-22:
https://github.com/YosysHQ/yosys/commit/f57524cf713e2cdc368ded8b7e13f85d310b8749 - fails

```
$ yosys --version
Yosys 0.9+2406 (git sha1 f57524cf, clang 6.0.0-1ubuntu2 -fPIC -Os)
```
```
make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port tests_tbx.w.data_in: \data_shadow <= \w.data_in
```

whitequark, 2020-06-22:
Huh, that's weird--I assumed it was a recent change that I introduced that broke your flow, but seems like it wasn't. Can you try a commit from a few months back? I'm not really sure when exactly it broke then...

davidthings, 2020-06-22:
Yep

https://github.com/YosysHQ/yosys/commit/f88bef767263590c94e157d0989afa91db3ccdb0 - fails

```
$ yosys --version
Yosys 0.9+2406 (git sha1 f88bef76, clang 6.0.0-1ubuntu2 -fPIC -Os)
```

```
$ make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port tests_tbx.w.data_in: \data_shadow <= \w.data_in
Makefile:30: recipe for target 'tests_tbx.json' failed
make: *** [tests_tbx.json] Error 1
```

daveshah1, 2020-06-22:
could it have been #1721 that broke this?

davidthings, 2020-06-22:
I'll check that next, Dave.  Meanwhile...

April 24 - https://github.com/YosysHQ/yosys/commit/bbf343589ba2b4508afdfaed17dfd11c3252cf7f - fails
```
$ yosys --version
Yosys 0.9+2406 (git sha1 bbf34358, clang 6.0.0-1ubuntu2 -fPIC -Os)
```

```
$ make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port tests_tbx.w.data_in: \data_shadow <= \w.data_in
Makefile:30: recipe for target 'tests_tbx.json' failed
make: *** [tests_tbx.json] Error 1
```

davidthings, 2020-06-22:
I'll check the one prior.

Feb 25 - https://github.com/YosysHQ/yosys/pull/1721/commit/5cae9c6e1628db7223a205d2444ef26e682007a5 - fails

```
$ yosys --version
Yosys 0.9+1706 (git sha1 5cae9c6e, clang 6.0.0-1ubuntu2 -fPIC -Os)
```
```
$ make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port tests_tbx.w.data_in: \data_shadow <= \w.data_in
Makefile:30: recipe for target 'tests_tbx.json' failed
make: *** [tests_tbx.json] Error 1
```

davidthings, 2020-06-22:
Uh oh...

Feb 22 - #1715 - https://github.com/YosysHQ/yosys/commit/6edca05793197a846bbfb0329e836c87fa5aabb6 - fails

```
$ yosys --version
Yosys 0.9+1706 (git sha1 6edca057, clang 6.0.0-1ubuntu2 -fPIC -Os)
```
```
$ make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port tests_tbx.w.data_in: \data_shadow <= \w.data_in
Makefile:30: recipe for target 'tests_tbx.json' failed
make: *** [tests_tbx.json] Error 1
```

whitequark, 2020-06-22:
Maybe check 0.9 next?

davidthings, 2020-06-22:
OK.  0.9 next.  Meanwhile...

 Jan 30 - https://github.com/YosysHQ/yosys/commit/60876ce183ee5f3980c378e190d996453d59a780 - fails

```
$ yosys --version
Yosys 0.9+1706 (git sha1 6edca057, clang 6.0.0-1ubuntu2 -fPIC -Os)
```

```
$ make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port tests_tbx.w.data_in: \data_shadow <= \w.data_in
Makefile:30: recipe for target 'tests_tbx.json' failed
make: *** [tests_tbx.json] Error 1
```

davidthings, 2020-06-22:
When was "0.9"?

whitequark, 2020-06-22:
1979e0b1f2482dbf0562f5116ab444280a377773

davidthings, 2020-06-22:
Dec 31 - https://github.com/YosysHQ/yosys/commit/c082329af33cd428f53f5afbcb51fab8de545090 - fails

```
$ yosys -V
Yosys 0.9+932 (git sha1 c082329a, clang 6.0.0-1ubuntu2 -fPIC -Os)
```
```
$ make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port tests_tbx.w.data_in: \data_shadow <= \w.data_in
Makefile:30: recipe for target 'tests_tbx.json' failed
make: *** [tests_tbx.json] Error 1
```

davidthings, 2020-06-22:
Something weird is going on. The test case fails here, but my actual code does compile.  I suspected it would because I know I have pulled Yosys since Nov 1.  

Nov 1 - https://github.com/YosysHQ/yosys/commit/c4bd318e76240d3e6a95109c19641cdfd86517b8 - fails

```
$ yosys -V
Yosys 0.9+932 (git sha1 c4bd318e, clang 6.0.0-1ubuntu2 -fPIC -Os)
```
```
$ make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port tests_tbx.w.data_in: \data_shadow <= \w.data_in
Makefile:30: recipe for target 'tests_tbx.json' failed
make: *** [tests_tbx.json] Error 1
```

whitequark, 2020-06-22:
I think the message is actually correct--you *are* assigning to inout bits that are used as outputs in the inner module.

davidthings, 2020-06-22:
OK.  Because you're you, we're going to assume you're correct...  

``` verilog
module worker(
        inout  [8:0] data_in,
        inout  [8:0] data_out
    );
    assign data_out[7:0] = data_in[7:0];
    assign data_in[ 8 ] = data_out[ 8 ];
endmodule

module tests_tbx (
        inout [8:0] data
    );
    wire [8:0] data_shadow;
    assign data_shadow[7:0] = 8'H55;
    worker w( data_shadow, data );
endmodule
``` 
But help me to understand... in tests_tbx, we assign data_shadow[7:0] with a thing, we pass that as param 1 - `data_in` into `worker`.  In `worker` we assign `data_out` from `data_in`.  How is that assigning to bits that are outputs?  We make no commitments about bit 8.

Meanwhile, I'm trying the bisection again with my actual code.  It still fails with HEAD but it succeeds with https://github.com/YosysHQ/yosys/commit/f57524cf713e2cdc368ded8b7e13f85d310b8749

whitequark, 2020-06-22:
> OK. Because you're you, we're going to assume you're correct...

That's a mistake... I was in fact wrong, I misread the code. Sorry for the noise.

davidthings, 2020-06-23:
@whitequark, it seems like it's  

4ef9ee3c4210a9d0c3b6491a0c89ee3365c77779 Jun 8 18:24

that breaks my code.  

```
ERROR: Mismatch in directionality for cell port usb_ad5592r_net_tbx.p_esc.escape_pipe_in: \escape_pipe_in <= \p_esc.escape_pipe_in
```
Whereas its immediate predecessor, 

467152d79fd4c8e25b48481f742fe318b1395728 Jun 8 17:38

Compiles it fine.

Is this enough information?  My "excellent" test code above fails in both cases, so it's not very diagnostic.  If you look at the check in, and nothing pops out I can have another go at refining a test case that mirrors the situation a little better.  Meanwhile... I have a recentish build that works!!

whitequark, 2020-06-23:
That... makes no sense. There's no actual changes in that commit outside of CXXRTL.

davidthings, 2020-06-23:
Quintuple checking

```
4e7d8377471c6e115cc80d97866fb41ff40c5835 - FAILS
4ef9ee3c4210a9d0c3b6491a0c89ee3365c77779 - FAILS
467152d79fd4c8e25b48481f742fe318b1395728 - WORKS
f2d7a18756088f4167b91c5ec4735bb694f36567 - WORKS
```

What would be awesome is if there were a simple test case that also fails in this way.  The right thing to do is to just leave it - there is a working build, but there is another deeper bug that it may be related to.  Let me work on it for a bit.

Thanks for your time so far @whitequark

davidthings, 2020-06-23:
Here's a tiny test that shows the error between the two builds.  

``` verilog 
module subworker(
        inout  [8:0] pipe_in,
        inout  [8:0] pipe_out
    );
    assign pipe_out[7:0] = pipe_in[7:0];
    assign pipe_in[ 8 ] = pipe_out[ 8 ];
endmodule

module worker (
        inout [8:0] pipe
    );
    wire [8:0] pipe_shadow;
    assign pipe_shadow[7:0] = 8'H55;
    subworker sw( pipe_shadow, pipe );
    assign pipe[8] = 1'B1;
endmodule

module tests_tbx (
        output [7:0] data,
        output      control
    );

    wire [8:0] pipe;

    assign data = pipe[7:0];
    assign control = pipe[ 8 ];

    worker w( pipe );
endmodule
```

Then here's what building with the two branches results in

```
$ yosys --version
Yosys 0.9+2406 (git sha1 467152d7, clang 6.0.0-1ubuntu2 -fPIC -Os)
$ make clean
rm -f tests_tbx.json tests_tbx.asc tests_tbx.rpt tests_tbx.bin pll.v
$ make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
nextpnr-ice40 --lp8k --freq 48 --top tests_tbx --package cm81 --pcf ../projects/tests_tbx/pins.pcf --json tests_tbx.json --asc tests_tbx.asc
$ # yosys has completed, no error
```

Switching to other toolset

```
$ yosys --version
Yosys 0.9+2406 (git sha1 4ef9ee3c, clang 6.0.0-1ubuntu2 -fPIC -Os)
$ make clean
rm -f tests_tbx.json tests_tbx.asc tests_tbx.rpt tests_tbx.bin pll.v
$ make
yosys -q -p 'synth_ice40 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port worker.sw.pipe_in: \pipe_shadow <= \sw.pipe_in
Makefile:30: recipe for target 'tests_tbx.json' failed
make: *** [tests_tbx.json] Error 1
```

whitequark, 2020-06-23:
Hm, now that I look closer, the Git history is confusingly nonlinear here. There are quite a few commits between the two you provided:
```
4ef9ee3c4 Merge pull request #2121 from whitequark/cxxrtl-debug-aliases
369dcb4e8 Merge pull request #2085 from rswarbrick/select
0f209378a Merge pull request #2089 from rswarbrick/modports
fbd0d8d5f Merge pull request #2105 from whitequark/split-flatten-off-techmap
f57524cf7 Merge pull request #2117 from PeterCrozier/struct_array
680913be8 Merge pull request #2119 from YosysHQ/mwk/fix-fsm-idstring
28b9f49c9 (origin/pr/2119) fsm_extract: avoid calling log_signal to determine wire name
76c499db7 (origin/pr/2117) Support packed arrays in struct/union.
5a5a9b4ff (origin/pr/2105) flatten: clean up log messages.
d731fe054 flatten: topologically sort modules.
6268bdfe6 flatten: simplify.
d3e210030 flatten: simplify. NFC.
66255dab4 flatten: simplify.
5d2b6d139 flatten: simplify. NFC.
3c3fa774e flatten: simplify. NFC.
e561a3a76 flatten: simplify. NFC.
678387680 flatten: simplify. NFC.
9338ff66b RTLIL: factor out RTLIL::Module::addMemory. NFC.
ebbbe2156 flatten: rename techmap-related stuff. NFC.
76c4ee4ea techmap, flatten: remove dead options.
6ac54a74f flatten: split from techmap.
d681c9df8 (origin/pr/2085) Pass some more args by reference in select.cc
061d1f0c0 Minor optimisations in select.cc's match_ids function
0d9beb5b2 Silence warning in select.cc
7746bba69 (origin/pr/2089) Simplify a modport check in hierarchy.cc
```
Can you narrow it down further?

davidthings, 2020-06-23:
You guys have been busy

davidthings, 2020-06-23:
I see!  Learning a bit here.  Despite being consecutive commits in the log, because the more recent one is a merge where your work is joined by the work of others, the actual issue could arise from anything between where you branched (210d129 @ 6/7/20) and 4ef9ee3.  Your list above is a list of those commits.  Perhaps I can narrow it down to one merge branch by testing the merges... 

[image]

davidthings, 2020-06-23:
Using the revised test code above (`module subworker`, etc.) , it looks like possibly d3e210030 is the culprit.

4ef9ee3c4 - **FAILS** - Merge pull request #2121 from whitequark/cxxrtl-debug-aliases   
369dcb4e8 - Merge pull request #2085 from rswarbrick/select   
0f209378a - Merge pull request #2089 from rswarbrick/modports   
fbd0d8d5f - **FAILS** - Merge pull request #2105 from whitequark/split-flatten-off-techmap   
---- 5a5a9b4ff (origin/pr/2105) flatten: clean up log messages.   
---- d731fe054 flatten: topologically sort modules.   
---- 6268bdfe6 - **FAILS** flatten: simplify.   
---- d3e210030 - **FAILS** flatten: simplify. NFC.      
---- 66255dab4 - **WORKS** - flatten: simplify.   
---- 5d2b6d139 - **WORKS** - flatten: simplify. NFC.   
---- 3c3fa774e flatten: simplify. NFC.   
---- e561a3a76 flatten: simplify. NFC.   
---- 678387680 flatten: simplify. NFC.   
---- 9338ff66b RTLIL: factor out RTLIL::Module::addMemory. NFC.   
---- ebbbe2156 flatten: rename techmap-related stuff. NFC.   
---- 76c4ee4ea techmap, flatten: remove dead options.   
---- 6ac54a74f flatten: split from techmap   
f57524cf7 - **WORKS** - Merge pull request #2117 from PeterCrozier/struct_array   
680913be8 - Merge pull request #2119 from YosysHQ/mwk/fix-fsm-idstring   
210d129d9 - **WORKS** - Merge pull request #2116 from whitequark/cxxrtl-vcd

whitequark, 2020-06-24:
Ah so looks like d3e2100 is the issue, I expected something like that.

davidthings, 2020-08-24:
Any progress on this @whitequark ?  I just encountered it again.  Please let me know if there's anything I can do to help.

whitequark, 2020-08-24:
Sorry, I completely forgot about this. I'll take a look.

whitequark, 2020-08-26:
@davidthings I just built commit 66255dab4 and it fails with the same error, so I think https://github.com/YosysHQ/yosys/issues/2183#issuecomment-648196622 is erroneous.

whitequark, 2020-08-26:
Minimized testcase:
```verilog
module foo(inout a, b);
  assign a = b;
endmodule
module bar(output c);
  foo f(c, 1'b0);
endmodule
```
The current `flatten` code actually swaps `a` and `b` and connects them the wrong way around, which is a pretty alarming bug.

whitequark, 2020-08-26:
Should be fixed in https://github.com/YosysHQ/yosys/pull/2356.

davidthings, 2020-08-26:
Thanks, @whitequark !

whitequark, 2020-08-26:
Can you check whether the PR fixes your original testcase?

davidthings, 2020-08-26:
Looks good.  Thanks again.

**Test Case**

``` verilog
module subworker(
        inout  [8:0] pipe_in,
        inout  [8:0] pipe_out
    );
    assign pipe_out[7:0] = pipe_in[7:0];
    assign pipe_in[ 8 ] = pipe_out[ 8 ];
endmodule

module worker (
        inout [8:0] pipe
    );
    wire [8:0] pipe_shadow;
    assign pipe_shadow[7:0] = 8'H55;
    subworker sw( pipe_shadow, pipe );
    assign pipe[8] = 1'B1;
endmodule

module tests_tbx (
        output [7:0] data,
        output      control
    );

    wire [8:0] pipe;

    assign data = pipe[7:0];
    assign control = pipe[ 8 ];

    worker w( pipe );
endmodule
```
**Using present yosys master**

$ yosys --version
Yosys 0.9+3521 (git sha1 4f2b78e1, clang 6.0.0-1ubuntu2 -fPIC -Os)

$ make
yosys -q -p 'synth_ice40 -abc9 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
ERROR: Mismatch in directionality for cell port worker.sw.pipe_in: \pipe_shadow <= \sw.pipe_in

**Using master + PR 2356**

$ yosys --version
Yosys 0.9+3521 (git sha1 9f089215, clang 6.0.0-1ubuntu2 -fPIC -Os)

$make
yosys -q -p 'synth_ice40 -abc9 -top tests_tbx -json tests_tbx.json' ../projects/tests_tbx/tests_tbx.v
\# no errors

PR #2356 (2020-08-27): flatten, techmap: don't canonicalize tpl driven bits via sigmap
For connection `assign a = b;`, `sigmap(a)` returns `b`. This is
exactly the opposite of the desired canonicalization for driven bits.
Consider the following code:

    module foo(inout a, b);
      assign a = b;
    endmodule
    module bar(output c);
      foo f(c, 1'b0);
    endmodule

Before this commit, the inout ports would be swapped after flattening
(and cause a crash while attempting to drive a constant value).

This issue was introduced in 9f772eb9.

Fixes #2183.
commit 9f089215 (2020-08-27): flatten, techmap: don't canonicalize tpl driven bits via sigmap.

For connection `assign a = b;`, `sigmap(a)` returns `b`. This is
exactly the opposite of the desired canonicalization for driven bits.
Consider the following code:

    module foo(inout a, b);
      assign a = b;
    endmodule
    module bar(output c);
      foo f(c, 1'b0);
    endmodule

Before this commit, the inout ports would be swapped after flattening
(and cause a crash while attempting to drive a constant value).

This issue was introduced in 9f772eb9.

Fixes #2183.
closed 2020-08-27 by whitequark