YosysHQ/yosys #2187: package import issue

gkamendje, 2020-06-24:
## Steps to reproduce the issue
In the file my_pkg.sv I have the following

package my_pkg;
       parameter KL=64;
endpackage

in the file top.sv I have the following
import my_pkg::*;
module top(
        input logic i_clk,
        output logic o_done
);
endmodule

## Expected behavior
The tool should accept the top level file since it is a valid syntax. This is a common way to import all the content of a package inside a module and hence avoid having to prefix the package name every time a package parameter is used inside the module. This works in Modelsim, VCS as well as in DC. Is there a way to accomplish this in Yosys?
## Actual behavior

top.sv:1: ERROR: syntax error, unexpected TOK_ID

zachjs, 2020-08-01:
You may want to give https://github.com/zachjs/sv2v a shot. It includes support for items at the top level, including imports.

gkamendje, 2020-08-01:
@zachjs  Thanks a lot for the hint. I tried it and it works great so far! I am leaving this issue open hoping that the YOSYS team will integrate top level imports in the near future.

nakengelhardt, 2024-03-25:
duplicate of #4006

closed 2024-03-25 by nakengelhardt