YosysHQ/yosys #2190: Stack overflow in XAigerWriter::bit2aig(Yosys::RTLIL::SigBit)

HackerFoo, 2020-06-24:
## Details

```
* thread #1, name = 'yosys', stop reason = signal SIGSEGV: invalid address (fault address: 0x7ffff80feff8)
    frame #0: 0x0000000000a643dd yosys`Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::SigBit, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::count(Yosys::RTLIL::SigBit const&) const + 29
yosys`Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::SigBit, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::count:
->  0xa643dd <+29>: callq  0x5d6998                  ; Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::SigBit, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::do_hash(Yosys::RTLIL::SigBit const&) const
    0xa643e2 <+34>: movq   0x8(%rsp), %rsi
    0xa643e7 <+39>: movq   (%rsp), %rdi
    0xa643eb <+43>: leaq   0x14(%rsp), %rdx

Backtrace:
* thread #1, name = 'yosys', stop reason = signal SIGSEGV: invalid address (fault address: 0x7ffff80feff8)
  * frame #0: 0x0000000000a643dd yosys`Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::SigBit, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::count(Yosys::RTLIL::SigBit const&) const + 29
    frame #1: 0x0000000000a66eb2 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 110
    frame #2: 0x0000000000a66f5d yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 281
    frame #3: 0x0000000000a66ed0 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 140
    frame #4: 0x0000000000a66f72 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 302
    frame #5: 0x0000000000a66ed0 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 140
    frame #6: 0x0000000000a66f5d yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 281
    frame #7: 0x0000000000a66ed0 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 140
    frame #8: 0x0000000000a66f5d yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 281
    frame #9: 0x0000000000a66ed0 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 140
    frame #10: 0x0000000000a66f5d yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 281
    frame #11: 0x0000000000a66ed0 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 140
    frame #12: 0x0000000000a66f5d yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 281
    frame #13: 0x0000000000a66f5d yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 281
    frame #14: 0x0000000000a66ed0 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 140
    frame #15: 0x0000000000a66f5d yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 281
    frame #16: 0x0000000000a66ed0 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 140
    frame #17: 0x0000000000a66f5d yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 281
    frame #18: 0x0000000000a66ed0 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 140
    frame #19: 0x0000000000a66f72 yosys`(anonymous namespace)::XAigerWriter::bit2aig(Yosys::RTLIL::SigBit) + 302
... repeating
```

## Steps to reproduce the issue
Unpack the tarball and run `sh crash.sh`

[crash.tar.gz](https://github.com/YosysHQ/yosys/files/4827512/crash.tar.gz)

Crashes with https://github.com/YosysHQ/yosys/commit/0835a86e30fc2a934f5e6c96b28c90b59654ed92, doesn't crash with yosys-2020.02.07 from [nixpkgs](https://nixos.org/nixos/packages.html?attr=yosys&channel=nixos-20.03&query=yosys) (https://github.com/YosysHQ/yosys/commit/2e8d6ec0b06b4e51e222c15c8049130bc264ae57).

Tested on Linux (NixOS 20.03) and macOS (10.14.6).

Ravenslofty, 2020-06-24:
cc @eddiehung

eddiehung, 2020-07-08:
I've spent a little time on this and didn't make much progress: it looks like `bit2aig()` is getting inside some combinatorial loop ending in a stack overflow. Please put some effort into reducing the testcase before I can look at it again.

eddiehung, 2020-08-03:
Please give #2325 a try. That fix works for me.

PR #2325 (2020-09-23): xilinx: do not make DSP48E1 a whitebox for ABC9 by default
Since it is a big fat block with little gains expected. Can still be enabled using a macro (in the attached testcase, DSP48E1 is ~8000 AIG nodes, ~18000 wires, 263 inputs and 154 outputs).

And make sure it works when it is enabled by fixing combinatorial loops cause problems in ABC9. Passes `techlibs/xilinx/test_dsp_models.h`.

Should fix #2190.
closed 2020-09-23 by eddiehung