YosysHQ/yosys #2191: Error: ABC execution of command "(...)/yosys-abc -s -f  (...)/abc.script 2>&1" failed: return code -1

zacchaeuswilliams2, 2020-06-24:
## Background

I am working with the [OpenROAD-flow](https://github.com/The-OpenROAD-Project/OpenROAD-flow) toolchain, and am trying to get [designs](https://github.com/mit-ll/CEP/tree/master/hdl_cores) from MIT LL Common Evaluation platform (CEP) to go through it. The toolchain use yosys for synthesis and makes use of the ABC tool.  I have been able to get the example designs and other designs from the CEP to go through the toolchain. With one [design](https://github.com/mit-ll/CEP/tree/master/hdl_cores/gps) in particular I am getting an esoteric error and the tools really don’t tell me what is going wrong. I am getting the same error as another [issue](https://github.com/YosysHQ/yosys/issues/1028) that was closed on account of the poster not giving enough information.

## Steps to reproduce the issue
Get the [gps](https://github.com/mit-ll/CEP/tree/master/hdl_cores/gps) design files from the CEP github (It also uses modules from the [AES](https://github.com/mit-ll/CEP/tree/master/hdl_cores/aes) directory as well) and try to synthesize them using the synthesis script [synth.tcl.](https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/master/flow/scripts/synth.tcl)

## Expected behavior

Nonzero error code is returned.

## Actual behavior

Error code of -1 is returned.

[image]

If I run this exact command that failed in terminal then it will run fine.

[image]

## Personal trouble shooting

I spent a while trying to figure out what was going wrong with this design and found some very interesting things. If I comment out a specific module in the design then this error does not happen. Also, I know specifically which line in the synth.tcl script the error is occurring from. It is occurring from the synth command on line 55 of the [script](https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/master/flow/scripts/synth.tcl). If I add the -noabc flag to this line then the error no longer occurs.

The S4 module that is causing this error does not occur when the S4 module is moved up the hierarchy as well.

The modules that are highlighted are the problem area. The problem is not with the design itself as I was able to get it to synthesize fine in Vivado.

[image]

Any input is highly appreciated, and if there is any information that I need to add please let me know.

Ravenslofty, 2020-06-24:
This is a downstream bug, you should file this at https://github.com/The-OpenROAD-Project/yosys instead.

daveshah1, 2020-06-24:
And if the problem isn't in OpenROAD's Yosys fork then it is almost certainly in ABC. Have you tried running abc inside valgrind to see if it is a memory issue that is only triggered when ABC is run by Yosys for some obscure, probably OS and libc-dependent (and therefore hard to reproduce) reason.

zacchaeuswilliams2, 2020-07-06:
Hello I tried seeing if Valgrind offered anymore information. There were no memory leaks that occurred. I've been spinning my wheels for some time now. Are there are other possible reasons this could be happening? I linked in the original issue that another person posted an issue about a similar problem months ago

whitequark, 2020-07-06:
This doesn't look like the same crash as #1028. In the issue you mention, abc crashed with a specific assertion. In this issue it just crashes. The only common thing is the crash.

I see that abc is allocating a lot (`10339896memKB`). Is it running out of memory?

zacchaeuswilliams2, 2020-07-06:
I am sure that it is not running out of memory as I am able to run this entire toolchain on larger designs and it completes without an error. That's a good point about the specific assertion failure. Is there a way that I can extract that information from the tool?

whitequark, 2020-07-07:
Try grabbing a backtrace with gdb.

wuxx, 2020-07-25:
I encounter the same problem when build for ULX3S with LiteX, under ubuntu 18.04 with the latest Yosys
[build fail for ULX3S](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/143)

-------
ubuntu@ubuntu:~/oss/ulx3s/linux-on-litex-vexriscv$ ./make.py --board=ulx3s --build
........
4.47.18.5. Executing ABC9.
ulx3s.txt

Running ABC command: /yosys-abc -s -f /abc.script 2>&1
ERROR: ABC: execution of command "/usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-ONKnaB/abc.script 2>&1" failed: return code -1.
Traceback (most recent call last):
File "./make.py", line 486, in
main()
File "./make.py", line 468, in main
builder.build(build_name="top", run=args.build)
File "/home/ubuntu/oss/ulx3s/litex/litex/soc/integration/builder.py", line 205, in build
vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
File "/home/ubuntu/oss/ulx3s/litex/litex/soc/integration/soc.py", line 1017, in build
return self.platform.build(self, *args, **kwargs)
File "/home/ubuntu/oss/ulx3s/litex/litex/build/lattice/platform.py", line 34, in build
return self.toolchain.build(self, *args, **kwargs)
File "/home/ubuntu/oss/ulx3s/litex/litex/build/lattice/trellis.py", line 229, in build
_run_script(script)
File "/home/ubuntu/oss/ulx3s/litex/litex/build/lattice/trellis.py", line 160, in _run_script
raise OSError("Subprocess failed")
OSError: Subprocess failed

ubuntu@ubuntu:~/oss/ulx3s/linux-on-litex-vexriscv$ /usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-ONKnaB/abc.script 2>&1
ABC command line: "source /tmp/yosys-abc-ONKnaB/abc.script".

read_lut /tmp/yosys-abc-ONKnaB/input.lut
read_box /tmp/yosys-abc-ONKnaB/input.box
&read /tmp/yosys-abc-ONKnaB/input.xaig
&ps
/tmp/yosys-abc-ONKnaB/input : i/o = 4562/ 5511 and = 20511 lev = 43 (7.14) mem = 0.48 MB box = 518 bb = 120
&scorr
Warning: The network is combinational.
&sweep
&dc2
&dch -f
&ps
/tmp/yosys-abc-ONKnaB/input : i/o = 4562/ 5511 and = 28653 lev = 35 (4.15) mem = 0.57 MB ch = 3686 box = 512 bb = 114
&if -W 300 -v
K = 7. Memory (bytes): Truth = 0. Cut = 60. Obj = 140. Set = 636. CutMin = no
Node = 28653. Ch = 3142. Total mem = 6.60 MB. Peak cut mem = 0.50 MB.
P: Del = 5041.00. Ar = 26111.0. Edge = 32699. Cut = 318202. T = 0.07 sec
P: Del = 4920.00. Ar = 25448.0. Edge = 31858. Cut = 316355. T = 0.07 sec
P: Del = 4920.00. Ar = 10437.0. Edge = 24062. Cut = 679900. T = 0.13 sec
F: Del = 4917.00. Ar = 7484.0. Edge = 21303. Cut = 582797. T = 0.13 sec
A: Del = 4917.00. Ar = 6933.0. Edge = 19344. Cut = 570621. T = 0.19 sec
A: Del = 4917.00. Ar = 6853.0. Edge = 19107. Cut = 593313. T = 0.20 sec
Total time = 0.79 sec
&write -n /tmp/yosys-abc-ONKnaB/output.aig
&mfs
&ps -l
/tmp/yosys-abc-ONKnaB/input : i/o = 4562/ 5511 and = 18843 lev = 27 (4.48) mem = 0.46 MB box = 509 bb = 114
Mapping (K=7) : lut = 5169 edge = 18760 lev = 10 (2.34) levB = 33 mem = 0.24 MB
LUT = 5169 : 2=836 16.2 % 3=1401 27.1 % 4=2017 39.0 % 5=755 14.6 % 6=78 1.5 % 7=82 1.6 % Ave = 3.63
&write -n /tmp/yosys-abc-ONKnaB/output.aig
time
elapse: 5.22 seconds, total: 5.22 seconds
ubuntu@ubuntu:~/oss/ulx3s/linux-on-litex-vexriscv$ echo $?
0

Ravenslofty, 2020-07-25:
The log you provided isn't very helpful, since ABC succeeds in it; plus it's under the ABC9 pass, not the ABC pass.

widlarizer, 2025-03-13:
Looks like there's nothing to do in relation to this bug report

closed 2025-03-13 by widlarizer