YosysHQ/yosys #2194: Sim pass do not support $memwr

mmicko, 2020-06-26:
After change https://github.com/YosysHQ/yosys/commit/abf81c76396d674833d1dce3f3f75a43f76e2a1f for memories now we get $memwr instead of $mem in example provided, and $memwr is unsupported, so test breaks.

Please find case in :
[test.zip](https://github.com/YosysHQ/yosys/files/4836534/test.zip)

mmicko, 2020-06-26:
@daveshah1 can you please check if memory handling is a special case for your change or this is something that is valid so $memwr needs to be supported ?

daveshah1, 2020-06-26:
I don't think I understand how that patch is creating `$memwr`s instead of `$mem`?

daveshah1, 2020-06-26:
Looking at the script there is no call to `memory_collect` or any other memory pass so I don't see how you could have ever had `$mem`, either.

mmicko, 2020-06-26:
Ok, thanks for checking, will ping @clairexen for this

daveshah1, 2020-06-26:
If it worked before the patch, maybe it was never trying to evaluate the `$memwr` so just silently broken instead of noisily broken.

clairexen, 2020-06-26:
> If it worked before the patch, maybe it was never trying to evaluate the $memwr so just silently broken instead of noisily broken.

that sounds like a reasonable guess to me.

mmicko, 2020-06-29:
From techlibs/common/simlib.v  
```
 Note that memory can only be simulated when all $memrd and $memwr cells
 have been merged to stand-alone $mem cells (this is what the "memory_collect"
 pass is doing).
```
So even when doing simulation with external tools it is assumed that $memwr and $memrd are not supported.
And when I do execute memory_collect before sim, all works as expected.

To make this consistent I would propose adding log_error when $memwr and $memrd are detected while executing sim command asking for memory_collect to be executed first and would updates tests accordingly.

whitequark, 2020-06-29:
> From techlibs/common/simlib.v

This only applies to Verilog, Yosys `sim` pass does not in principle have this restriction.

mmicko, 2020-06-30:
We will probably touch this again when we cleanup memory interface, but it fixed current issues

closed 2020-06-30 by mmicko