YosysHQ/yosys #2198: test_pmgen -generate xilinx_srl.fixed failing

mmicko, 2020-06-29:
Due to change in https://github.com/YosysHQ/yosys/commit/86ab7d3a6ed994c06654fbae9770caa030f115d6 executing `test_pmgen -generate xilinx_srl.fixed` will fail with error trying to get non-existing parameter values for IS_C_INVERTED, IS_R_INVERTED and IS_D_INVERTED.

It is possible to fix crash and make test pass by change instance to something like `param(next, \IS_C_INVERTED, false)` but is it a proper fix ?

mwkmwkmwk, 2020-06-29:
Yes, it is — some FF types (particularly the `_1` types) just don't have the inversion parameters...

mwkmwkmwk, 2020-06-29:
So this doesn't seem to be a `xilinx_srl` bug to me — the problem here is that `xilinx_srl` expects the `+/xilinx/cells_sim.v` file to be loaded, so that the default parameter values are resolved automatically.  Running `read_verilog -lib +/xilinx/cells_sim.v` before running `test_pmgen` fixes the problem.

Perhaps we need a documentation fix instead?

mmicko, 2020-06-29:
Thanks. Added this into proper yosys-tests test case already

mwkmwkmwk, 2020-07-13:
This can be closed, right?

mmicko, 2020-07-13:
All fine and confirmed by yosys-test

closed 2020-07-13 by mmicko