YosysHQ/yosys #2204: Techmap pass makes attributes disappear

acomodi, 2020-06-29:
## Steps to reproduce the issue

[techmap_attr_failure.tar.gz](https://github.com/YosysHQ/yosys/files/4847259/techmap_attr_failure.tar.gz)
1. Build and install yosys (https://github.com/YosysHQ/yosys/commit/4160acc0b19532d50435abbc66bd2f6aab9473b6)
1. Unpack the tar
1. `cd techmap_attr && make`

## Expected behavior

The Makefile checks whether the LOC constraint on the primitive cell is present both in the pre- and post- mapped verilog output. The LOC attribute (or any custom attribute) should not get lost.

## Actual behavior

The LOC attribute gets lost in the post-mapped output.

## Insights

I have run a `git bisect` and found the first "bad commit" being 76c4ee4e

whitequark, 2020-06-29:
Reproduced and confirmed the causative commit. Thanks for the great test case and bisection!

whitequark, 2020-06-29:
Fixed in https://github.com/YosysHQ/yosys/pull/2205

closed 2020-06-30 by whitequark

acomodi, 2020-06-30:
@whitequark Great, thanks for the quick fix! I can confirm it works fine with my original test from which I extracted the reduced one.

PR #2205 (2020-06-30): techmap: don't drop attributes on replaced cells
This was introduced in 76c4ee4ea5cb6a3dc214f66237af22a1bedda010.

Fixes #2204.
commit a97c13f0 (2020-06-30): techmap: don't drop attributes on replaced cells.

This was introduced in 76c4ee4ea5cb6a3dc214f66237af22a1bedda010.

Fixes #2204.