YosysHQ/yosys #2207: make bumpversion does not work on darwin

edbordin, 2020-06-30:
## Steps to reproduce the issue

1. Clone from master on OS X Catalina
2. Run `make bumpversion`

## Expected behavior

Makefile is updated to reflect current version relative to latest release.

## Actual behavior
```
sed -i "/^YOSYS_VER := / s/+[0-9][0-9]*$/+`git log --oneline 8a4c6e6.. | wc -l`/;" Makefile
sed: 1: "Makefile": invalid command code M`
```

BSD `sed` installed on OS X is not compatible with GNU sed. Similarly, BSD `wc` inserts unwanted whitespace.

## Suggested solutions

Homebrew has `gnu-sed` and `coreutils` packages providing GNU versions. Macports has `gsed` and `coreutils` packages. If you assume that one of these options is installed it would be enough to use `gsed` and `gwc` on Darwin. Alternatively you can add the GNU versions to the PATH under their standard names `sed` and `wc`. I'm happy to prepare a PR using either approach if someone can provide guidance.

whitequark, 2020-06-30:
Why are you running `make bumpversion`? This command is for upstream use only

edbordin, 2020-06-30:
I wanted to output a version string that nmigen understands while also updating it every day https://github.com/open-tool-forge/fpga-toolchain/issues/33

whitequark, 2020-06-30:
I see! Please do *not* change the part of the version after `+` in any way compared to the upstream packages, nMigen actually relies on it being the same as upstream and any changes would make my job harder.

edbordin, 2020-06-30:
I'll close this now that I have a better understanding of what's going on.

closed 2020-06-30 by edbordin