YosysHQ/yosys #2212: Suggestions to Improve debug/visualisation in SMT2 backend for SBY

dh73, 2020-06-30:
## Steps to reproduce the issue

Consider this dummy test, Verilog and SBY files:
```verilog
`default_nettype none
module fsm (input wire clk, rstn, start, a, b, c, d,
            output reg [1:0] state_val);
        
        localparam [3:0] s0=0, s1=1, s2=2, s3=3;
        reg [1:0] ps, ns;

        wire jump_s2, jump_s3, return_w;
        assign jump_s2 = &({a,b,c,d});
        assign jump_s3 = (~a & b & c & ~d);
        assign return_w  = (a | d & ~b);

        always @(posedge clk) begin
                if (~rstn) ps <= s0;
                else       ps <= ns;
        end
        
        always @(*) begin
                ns = ps;
                case (ps)
                        s0: if (start)   ns = s1;
                        s1: if (jump_s2) ns = s2;
                        s2: if (jump_s3) ns = s3;
                        s3: if (return_w)  ns = s0;
                endcase
        end
        assign state_val = ps;

        // Assertion logic
        always @(*) assume (rstn == 1'b1);
        always @(posedge clk) if (ps == s2 && jump_s3) assert (ps == s3);
endmodule
```
SBY:
```
[options]
mode bmc

[engines]
smtbmc boolector

[script]
read_verilog -formal fsm.v
prep -top fsm

[files]
fsm.v
```
## Expected behavior
When a formal tool finds a failure, it shows a concise CEX with logic that influences the assertion/cover logic. For this particular example the logic influences are: `ps`, `jump_s3`, `clk` and `rstn`, as shown in the following VCD. The rest of the tools also shows the exact moment where the assertion failed (see `assert_always_i` transition from 1 to 0 (falsification)).

[image]

## Actual behavior
User needs to get signals from the generated VCD file, and correlate it back with the RTL. For trivial modules it may not represent a problem. For big designs, both in size and depth (states), this kind of debuggability may represent a barrier of tool adoption. 

[image]

A quick workaround is to separate the formal "testbench" into separate file, `bind` or instantiate it into the top module, and when a CEX or reached cover is found, get the signals from the container, as starting point. But still requires human intervention. 

```verilog
// Assert container
module assert_always (input wire event_m, disable_e, precondition, condition);
`ifndef _SV_
	always @(*) assume (disable_e == 1'b1);
	always @(posedge event_m)
                if (precondition) assert (condition);
`else
	assert_always_i: assert property (@(posedge event_m) disable iff (!disable_e) 
					    precondition |-> condition);
`endif

// Assert container instance
wire cause  = (ps == s2 && jump_s3);
wire effect = (ps == s3); // check ns instead, to fix it
assert_always checker_inst (clk, rstn, cause, effect);
endmodule
```

Maybe a better approach is to add a mechanism in `backends/smt2/smt2.cc` to add that information in the SMT generated model, as a comment like how it extracts the assertion name, and let `smtio.py` and `smtbmc.py` deal with the extra information needed in the `generate_vcd` function.

nakengelhardt, 2020-07-01:
This is really more of a SymbiYosys issue than a Yosys issue; but there will probably need to be some handling on the Yosys side to enable it.

A first rough idea would be to select the fan-in cone of the property that was triggered, and create a `.gtkw` file with the public names from that cone already added to the waveform. Might need something fancier for big designs, as the entire cone might be too big, but a fixed number of levels might lead to odd results because of differing amounts of non-public logic inbetween.

dh73, 2020-07-01:
I can move it if that's better. I opened here because AFAICS the smt2 already dumps the private names of the wires, so it will be probably more easy to deal with this in the SMT backed that is hosted in Yosys, but IDK.