YosysHQ/yosys #2213: Removal of conditional code for ABC/ABC9

litghost, 2020-06-30:
In earlier versions of the ABC/ABC9 support in yosys, there was a defined `_ABC` or `_ABC9` that could be used to conditionally select code that is suitable for ABC/ABC9 versus other uses. 

`-D_ABC9` https://github.com/YosysHQ/yosys/blob/d4212d128b5985cf09f5e7f14bc06e7323e644ac/techlibs/xilinx/synth_xilinx.cc#L286-L292

or

`-D_ABC`
https://github.com/YosysHQ/yosys/blob/7959e9d6b25d7afefded4b14e14ccf2b0b5af553/techlibs/xilinx/synth_xilinx.cc#L286-L289

It appears these were removed in https://github.com/YosysHQ/yosys/commit/1dc22607c38486d9e1a2b56f749d1eca35d405d2

The question is, why we this removed?  In cases where conditional code is desired and/or needed for different needs (abc/abc9/simulation), what is the recommended way to support this?

eddiehung, 2020-07-01:
That macro was something that was necessary to work around some ABC9 modelling limitations. It's removal was because -- to my knowledge -- I had fixed those limitations and can now interpret the unmodified simulation model. Please be more direct about what the breakage is, or propose a new PR showing why they would still be necessary for your use case.

litghost, 2020-07-01:
You may not remember, but you are actually the one who suggested using `_ABC`, specifically for:

https://github.com/SymbiFlow/yosys/blob/6bccd35a41ab82f52f0688478310899cfec04e08/techlibs/xilinx/cells_sim.v#L279-L285

The underlying hardware in this case is not precisely modeled in either branch, as there is in fact a bitstream configured mux between CIN and CYINIT, rather than a logic `|`.  The last time we talked, you proposed the above code as a reasonable way to express the hardware.  Given that `_ABC` (or `_ABC9`) is now gone, this naturally no longer works.

eddiehung, 2020-07-08:
I honestly had completely forgotten about that. I would be in favour of putting it back in with a comment that explains its necessity -- go ahead and do so in a PR. It does feel a little fishy to detect unconnected ports by checking equality with `1'bz` though.

litghost, 2020-07-08:
> It does feel a little fishy to detect unconnected ports by checking equality with `1'bz` though.

I'm not sure verilog provides a superior method.  It is worth noting that Vivado's unisim uses a similiar construction: https://github.com/Xilinx/XilinxUnisimLibrary/blob/master/verilog/src/unisims/CARRY4.v#L73-L74   The unisim construction is actually a more robust construction, because it handles the "default" case, where both ports are disconnected.

I'm open to other suggestions for how to model this kind of construction.