YosysHQ/yosys #2221: Occasional failure of tests/fsm

ldoolitt, 2020-07-03:
## Steps to reproduce the issue

Build yosys (commit a9b61080) and abc (commit 83f54185) from source.
I have a weird process for this step that I'm 99% sure is not relevant to this bug.
`cd tests/fsm && bash run-test.sh "-S 4605826776001879497"`

## Expected behavior

(as seen with a different seed)
`generating tests..`
`PRNG seed: 6301643816686424051`
`running tests..`
`[0]K[1]K[2]K[3]K[4]K[5]T[6]K[7]K[8]K[9]K[10]K[11]K[12]K[13]K[14]K[15]K[16]K[17]K[18]K[19]K[20]K[21]K[22]K[23]K[24]K[25]K[26]K[27]K[28]K[29]K[30]K[31]K[32]K[33]K[34]K[35]K[36]K[37]K[38]K[39]K[40]K[41]K[42]K[43]K[44]K[45]K[46]K[47]K[48]K[49]K`

## Actual behavior

`generating tests..`
`PRNG seed: 4605826776001879497`
`running tests..`
`[0]K[1]K[2]K[3]K[4]K[5]K[6]K[7]K[8]K[9]K[10]K[11]K[12]K[13]K[14]K[15]K[16]K[17]K[18]K[19]K[20]K[21]K[22]K[23]K[24]K[25]K[26]K[27]K[28]K[29]Warning: Regarding the user-specified fsm_encoding attribute on gate.state:`
 `  Users of state reg look like FSM recoding might result in larger circuit.`
 `   Doesn't look like a proper FSM. Possible simulation-synthesis mismatch!`
`K[30]K[31]terminate called after throwing an instance of 'std::bad_alloc'`
`  what():  std::bad_alloc`
`make: *** [temp/makefile:160: temp/uut_00031.log] Aborted`

daveshah1, 2020-07-03:
Do you see anything suspicious if you run Yosys under valgrind?

ldoolitt, 2020-07-03:
I can reproduce the failure at will.
The environment is stock Debian Stable (Buster) on amd64, in a chroot.
Any hints on how to run this test under valgrind?  I'm sure I can figure it out in 20 minutes.

ldoolitt, 2020-07-03:
I got as far as
`$ ../../yosys -ql temp/uut_00031.out temp/uut_00031.ys`
`terminate called after throwing an instance of 'std::bad_alloc'`
`  what():  std::bad_alloc`
`Aborted`

daveshah1, 2020-07-03:
Try prefixing that command with `valgrind` and see if it reports any memory issues.

ldoolitt, 2020-07-03:
OK, done. I'm not practiced with valgrind, so I zipped up and attached all the relevant files (I hope).
Comparing it to a non-crashing test, I don't see any issues to blame.
Let me know if I can help more.
[bug_2221.zip](https://github.com/YosysHQ/yosys/files/4871831/bug_2221.zip)

Here it is under gdb:
`$ gdb ../../yosys`
`GNU gdb (Debian 8.2.1-2+b3) 8.2.1`
`Copyright (C) 2018 Free Software Foundation, Inc.`
`License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>`
`This is free software: you are free to change and redistribute it.`
`There is NO WARRANTY, to the extent permitted by law.`
`Type "show copying" and "show warranty" for details.`
`This GDB was configured as "x86_64-linux-gnu".`
`Type "show configuration" for configuration details.`
`For bug reporting instructions, please see:`
`<http://www.gnu.org/software/gdb/bugs/>.`
`Find the GDB manual and other documentation resources online at:`
`    <http://www.gnu.org/software/gdb/documentation/>.`
`For help, type "help".`
`Type "apropos word" to search for commands related to "word"...`
`Reading symbols from ../../yosys...done.`
`(gdb) run -ql temp/uut_00031.out temp/uut_00031.ys`
`Starting program: /home/builder/src/yosys-bin/yosys -ql temp/uut_00031.out temp/uut_00031.ys`
`[Thread debugging using libthread_db enabled]`
`Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".`
`terminate called after throwing an instance of 'std::bad_alloc'`
`  what():  std::bad_alloc`
`Program received signal SIGABRT, Aborted.`
`__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50`
`50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.`
`(gdb) bt`
`#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50`
`#1  0x00007ffff74c0535 in __GI_abort () at abort.c:79`
`#2  0x00007ffff7ecd983 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6`
`#3  0x00007ffff7ed38c6 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6`
`#4  0x00007ffff7ed3901 in std::terminate() ()`
`   from /lib/x86_64-linux-gnu/libstdc++.so.6`
`#5  0x00007ffff7ed3b89 in __cxa_rethrow ()`
`   from /lib/x86_64-linux-gnu/libstdc++.so.6`
`#6  0x00005555557156bb in Yosys::run_frontend (filename="temp/uut_00031.ys", `
`    command=..., backend_command=<optimized out>, `
`    from_to_label=<optimized out>, design=0x555555e93500)`
`    at /home/larry/git/yosys/kernel/yosys.cc:1003`
`#7  0x000055555569f3b5 in main (argc=4, argv=0x7fffffffeb58)`
`    at /usr/include/c++/8/ext/new_allocator.h:79`

ldoolitt, 2020-07-03:
I ran the test interactively, and I can tell you that the crash happens on the first `opt` step in uut_00031.ys.

daveshah1, 2020-07-04:
Thanks, I can reproduce a similar failure on my Arch Linux machine:
```
2.9. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Found and cleaned up 9 empty switches in `\uut_00031.$proc$temp/uut_00031.v:12$2'.
Removing empty process `uut_00031.$proc$temp/uut_00031.v:12$2'.
Cleaned up 9 empty switches.
Removed 0 unused cells and 22 unused wires.
Renaming module \uut_00031 to \gate.

3. Executing OPT pass (performing simple optimizations).

3.1. Executing OPT_EXPR pass (perform const folding).
Optimizing module gate.

Catchpoint 1 (exception thrown), 0x00007ffff7e2a062 in __cxxabiv1::__cxa_throw (obj=obj@entry=0x555555f63660, 
    tinfo=0x7ffff7f56250 <typeinfo for std::length_error>, dest=0x7ffff7e3ff50 <std::length_error::~length_error()>)
    at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:78
78	/build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc: No such file or directory.
(gdb) bt
#0  0x00007ffff7e2a062 in __cxxabiv1::__cxa_throw (obj=obj@entry=0x555555f63660, tinfo=0x7ffff7f56250 <typeinfo for std::length_error>, 
    dest=0x7ffff7e3ff50 <std::length_error::~length_error()>) at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:78
#1  0x00007ffff7e2055f in std::__throw_length_error (__s=__s@entry=0x555555c477c7 "cannot create std::vector larger than max_size()")
    at /build/gcc/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:78
#2  0x00005555556a6888 in std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::_S_check_init_len (__a=..., 
    __n=18446744073709551613) at /usr/include/c++/10.1.0/ext/new_allocator.h:89
#3  std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::_M_range_initialize<__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > > > (__last=..., __first=..., this=0x7fffffffc220)
    at /usr/include/c++/10.1.0/bits/stl_vector.h:1582
#4  std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::vector<__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, void> (__a=..., __last=
    {wire = 0x1b0, {data = 49, offset = 49}}, __first={wire = 0x12, {data = 49, offset = 49}}, this=0x7fffffffc220)
    at /usr/include/c++/10.1.0/bits/stl_vector.h:657
#5  Yosys::RTLIL::SigSpec::extract (this=0x7fffffffc220, this@entry=0x7fffffffc7f8, offset=offset@entry=29, length=<optimized out>)
    at kernel/rtlil.cc:3467
#6  0x000055555591d274 in Yosys::RTLIL::SigSpec::extract_end (this=this@entry=0x7fffffffc7f8, offset=offset@entry=29)
    at ./kernel/rtlil.h:871
#7  0x000055555590e524 in (anonymous namespace)::replace_const_cells (design=design@entry=0x555555e86440, 
    module=module@entry=0x555555eb58e0, consume_x=consume_x@entry=false, mux_undef=mux_undef@entry=false, mux_bool=mux_bool@entry=false, 
    do_fine=do_fine@entry=false, keepdc=false, clkinv=false) at passes/opt/opt_expr.cc:1578
#8  0x000055555591cf20 in (anonymous namespace)::OptExprPass::execute (this=this@entry=0x555555e23cc0 <(anonymous namespace)::OptExprPass>, 
    args=std::vector of length 1, capacity 1 = {...}, design=design@entry=0x555555e86440) at passes/opt/opt_expr.cc:2110
#9  0x000055555569816b in Yosys::Pass::call (design=design@entry=0x555555e86440, args=std::vector of length 1, capacity 1 = {...})
    at /usr/include/c++/10.1.0/bits/stl_vector.h:1043
#10 0x0000555555698531 in Yosys::Pass::call (design=design@entry=0x555555e86440, 
    command="p\363\365UUU\000\000\b\000\000\000\000\000\000\000opt_expr\000\377\377\377\000\000\000\000\300\"\361UUU\000\000\061\000\000\000\000\000\000\000\240\363\365UUU\000\000\b\000\000\000\000\000\000\000opt_expr\000\360\363UUU\000\000\320\360\363UUU\000\000q\000\000\000\000\000\000\000\377\377\377\377\000\000\000\000\377\377\377\377\377\377\377\377\001\000\000\000", '\377' <repeats 72 times>, "UU\000\000\240\030\361UUU\000\000"...) at kernel/register.cc:293
#11 0x00005555558c3642 in (anonymous namespace)::OptPass::execute (this=0x7fffffffd258, 
    this@entry=0x555555e22700 <(anonymous namespace)::OptPass>, 
    args=std::vector of length -4398046510741, capacity 257690079906735342 = {...}, design=design@entry=0x555555e86440)
    at passes/opt/opt.cc:156
#12 0x000055555569816b in Yosys::Pass::call (design=design@entry=0x555555e86440, args=std::vector of length 1, capacity 1 = {...})
    at /usr/include/c++/10.1.0/bits/stl_vector.h:1043
#13 0x0000555555698366 in Yosys::Pass::call (design=design@entry=0x555555e86440, 
    command="p\361\365UUU\000\000\003\000\000\000\000\000\000\000opt\000_00031 to \\gate.\n\000\000\000q\000\000\000\000\000\000\000", '\377' <repeats 52 times>, "\000\000\000\000", '\377' <repeats 36 times>, "\000\000\000\000\220\344\366UUU\000\000!\000\000\000\000\000\000\000p\362\365UUU\000\000 H\366UUU\000\000 \000\000\000\000\000\000\000\060\000\000\000\000\000\000\000\060\362\365UUU\000\000"...)
    at kernel/register.cc:270
#14 0x00005555556ec7d7 in Yosys::run_frontend (filename="temp/uut_00031.ys", command=..., backend_command=0x0, 
    from_to_label=0x7fffffffd538, design=0x555555e86440) at kernel/yosys.cc:988
#15 0x0000555555678630 in main (argc=2, argv=0x7fffffffd9f8) at /usr/include/c++/10.1.0/ext/new_allocator.h:79
(gdb)
```
(with `catch throw`)

ldoolitt, 2020-07-04:
I shunk the test case down.  See attached bug_2221.v.
[bug_2221.txt](https://github.com/YosysHQ/yosys/files/4873790/bug_2221.txt)
```
$ yosys bug_2221.v -p opt
[usual boilerplate]
 Yosys 0.9+2406 (git sha1 a9b61080, gcc 8.3.0-6 -fPIC -Os)
-- Parsing `bug_2221.v' using frontend `verilog' --
1. Executing Verilog-2005 frontend: bug_2221.v
Parsing Verilog input from `bug_2221.v' to AST representation.
Generating RTLIL representation for module `\uut_00031'.
Successfully finished Verilog frontend.
-- Running command `opt' --
2. Executing OPT pass (performing simple optimizations).
2.1. Executing OPT_EXPR pass (perform const folding).
Optimizing module uut_00031.
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted
```

mwkmwkmwk, 2020-07-04:
That's one of my screwups, I'm afraid...

ldoolitt, 2020-07-05:
I confirm that commit edc040bd fixes this problem for me
```
$ cd tests/fsm && bash run-test.sh "-S 4605826776001879497"
generating tests..
PRNG seed: 4605826776001879497
running tests..
[0]K[1]K[2]K[3]K[4]K[5]K[6]K[7]K[8]K[9]K[10]K[11]K[12]K[13]K[14]K[15]K[16]K[17]K[18]K[19]K[20]K[21]K[22]K[23]K[24]K[25]K[26]K[27]K[28]K[29]Warning: Regarding the user-specified fsm_encoding attribute on gate.state:
    Users of state reg look like FSM recoding might result in larger circuit.
    Doesn't look like a proper FSM. Possible simulation-synthesis mismatch!
K[30]K[31]K[32]K[33]K[34]K[35]K[36]K[37]K[38]K[39]K[40]K[41]K[42]K[43]K[44]K[45]K[46]K[47]K[48]K[49]K
```
I don't know the customs here.  Should I wait for the patch to show up in master before closing?

mwkmwkmwk, 2020-07-05:
You don't need to do anything, merging the PR will automatically close the issue.

PR #2231 (2020-07-05): opt_expr: Fix crash on $mul optimization with more zeros removed than…
… Y has.

Fixes #2221.
commit 7afcb72c (2020-07-05): opt_expr: Fix crash on $mul optimization with more zeros removed than Y has.

Fixes #2221.
closed 2020-07-05 by mwkmwkmwk