YosysHQ/yosys #2241: ERROR: Abort in frontends/ast/simplify.cc:4239.

Speccery, 2020-07-06:
## Steps to reproduce the issue

Current version of yosys reproduces the problem in the title with both ice40 and ecp5 targets. The error message is shown pretty much immediately.

```
git clone https://github.com/Speccery/icy99.git
make

yosys  -q -p "synth_ice40 -top top_blackice2 -abc2 -blif erik9900.blif" src/ram2.v src/sys.v src/rom.v src/tms9900.v src/alu9900.v src/tms9902.v src/erik_pll.v src/tms9918.v src/vga_sync.v src/dualport1k9.v src/dualport4k.v src/dualport512_9.v src/xmemctrl.v src/gromext.v src/serloader.v src/serial_rx.v src/serial_tx.v src/spi_slave.v src/tms9901.v src/lcd_sys.v lcd/pmodoledrgb_controller.v lcd/ram_source.v src/dualport_par.v src/ps2kb.v top_blackice2.v
Warning: Replacing memory \keyboard with list of registers. See src/sys.v:374, src/sys.v:361, src/sys.v:343
ERROR: Abort in frontends/ast/simplify.cc:4239.
Makefile:18: recipe for target 'erik9900.blif' failed
make: *** [erik9900.blif] Error 1
```

Sorry that I haven't minimized the code (yet). Will try to reduce code size later, but without context where the error originates it's a bit hard to know where to start reducing the code. The warning above occurs in my current code. 
Build from 2020-05-24 works, current version consistently produces the error above.

## Expected behavior
Build proceeds.
The build will stop prematurely as my git repository does not included the ROM memory initialization files currently (on purpose). 

## Actual behavior
**ERROR: Abort in frontends/ast/simplify.cc:4239.**

The error above is produced almost immediately.

whitequark, 2020-07-06:
Can you bisect Yosys?

whitequark, 2020-07-06:
Actually nevermind, this was likely introduced by 7ad0c49905d97c3bea2b74b76ca0feb87d21f70d. cc @clairexen

daveshah1, 2020-07-06:
Minimal testcase:

```verilog
module top(input [1:0] sel, output reg [3:0] q);
	always @(sel) begin
		q = 4'b0000;
		q[sel] = 1'b1;
	end
endmodule
```

It doesn't fail using the more common (and better practice) `always @*` instead of an explicity sensitivity list but that doesn't change the fact this is a Yosys bug (changing to @* in the design should be a workaround though)

Speccery, 2020-07-06:
Thanks @daveshah1 very much!

PR #2256 (2020-07-10): Add AST_EDGE support to AstNode::detect_latch()
Fixes #2241.
commit f9ed0942 (2020-07-10): Add AST_EDGE support to AstNode::detect_latch(), fixes #2241
closed 2020-07-10 by clairexen