YosysHQ/yosys #2250: Doubts about YosysJS

lcbcFoo, 2020-07-08:
Hi! 

(I am skipping the issue template since these are questions about YosysJS and not an issue.)

For context, I am developing a web application which currently requires Yosys to be installed in the server machine and client side requests commands through REST API which are executed by Yosys and then responded to the client within the REST response object. If possible, I am very interested in embedding Yosys in the client side application if possible (would make the application entirely client-sided, eliminating the server need).

I am puzzled if YosysJS could be used for that, since I don't understand exactly how it works and can't find enough information about it. Would greatly appreciate any clarifications for these points:

1 - YosysJS would completely remove the need to install Yosys in the server? Or it acts only as a Javascript API that runs Yosys in the background?
2 - In case it is indeed a replacement for the default Yosys installation and not just an API, how does it work? Yosys source code is built for Javascript? (This is more a personal curiosity point, seems like magic for me).
3 - What is the current state of YosysJS in terms of development/versions? The reason for this question is that I can find very few references for YosysJS or releases history.

Appreciate any help, thanks!

whitequark, 2020-07-08:
1. YosysJS is a fully featured build of Yosys.
2. YosysJS uses Emscripten to compile Yosys to JavaScript or Wasm.
3. It should in principle work on master but I suspect no one tested it recently. You could try building it and filing bugs for anything that breaks.

lcbcFoo, 2020-07-08:
Thanks for the very fast answer @whitequark 

Very interesting, I will give it a try and report back if it works or I find any issues. I never used Emscripten and I can't find specific mentions in the build steps to build YosysJS, are there any additional steps required or further references to follow?

whitequark, 2020-07-08:
Install emscripten, run `make config-emcc; emmake`

lcbcFoo, 2020-07-09:
I managed to build, but faced issues when running. I believe some incompatible problems exists with newer versions of emscripten. I will open a separate issue for that.
Closing.

closed 2020-07-09 by lcbcFoo