YosysHQ/yosys #2251: YosysJS error: ReferenceError: _memset is not defined

lcbcFoo, 2020-07-09:
## Steps to reproduce the issue

- Using latest release of emscripten (1.39.19) for build.
- Tried both with master and latest tagged release yosys-0.9.
- Build YosysJS with 
```
make config-emcc; emmake make
```
- Serve the generated code (tested with node's module `http-server` and python `python -m http.server`)
- Open any demo HTML page and try to run commands. What I exactly did was to navigate to demo02.html and try to run any of the synthesis buttons.

## Expected behavior

Working synthesis, similar behavior to http://www.clifford.at/yosys/nogit/YosysJS/snapshot/demo02.html

## Actual behavior

The following error happens. Full build and error logs attached.

```
yosyswrk.js:7 -- Running command `design -reset; read_verilog input.v; show -stretch' --
yosyswrk.js:7 ReferenceError: _memset is not defined
yosyswrk.js:7 ReferenceError: _memset is not defined
```

[build-esmcripten-latest.log](https://github.com/YosysHQ/yosys/files/4893850/build-esmcripten-latest.log)
[run-log-latest.txt](https://github.com/YosysHQ/yosys/files/4893851/run-log-latest.txt)

## Additional information

I tracked in the [emscripten release notes ](https://emscripten.org/docs/introducing_emscripten/release_notes.html) that in release `v1.39.11: 03/20/2020` they made what look like breaking changes related to _memset. However, when I tried to build YosysJS with an older version of emscripten (v1.39.0) I got a different error.

Since I don't know if this is indeed a compatibility problem with newer versions of emscripten and which would be the latest emscripten compatible version I decided to keep this issue report with the results I got using their latest version.

daveshah1, 2020-07-09:
As you have Emscripten set up, can you see if adding '_memset' to this list fixes it: https://github.com/YosysHQ/yosys/blob/master/Makefile#L250 ?

lcbcFoo, 2020-07-09:
@daveshah1 yeap, that was enough to fix it. I will make a PR with that change.
Thank you!

closed 2020-07-09 by clairexen

qarlosalberto, 2020-12-20:
@lcbcFoo could you share your built yosysjs?

commit 68babb2a (2020-07-09): Fix issue #2251 (#2252)

* Fix #2251 - YosysJS ReferenceError: _memset is not defined.
Add '_memset' in emcc EXPORTED_FUNCTIONS in Makefile.