YosysHQ/yosys #2253: Verilog parser now has conflicts

whitequark, 2020-07-09:
Since one of the recent PRs got merged, the parser now has both SR and RR conflicts:
```
frontends/verilog/verilog_parser.y: warning: 3 shift/reduce conflicts [-Wconflicts-sr]
frontends/verilog/verilog_parser.y: warning: 4 reduce/reduce conflicts [-Wconflicts-rr]
```

whitequark, 2020-07-09:
Looks like 7450ee7f8 (#2203) introduces 3 S/R conflicts.

whitequark, 2020-07-09:
Yeah and 802671b22 (#2244) introduces 4 R/R conflicts.

whitequark, 2020-07-09:
@kgugala Please fix the conflicts or I'd have to revert your changes.

@clairexen Let's turn unexpected yacc conflicts into CI failures to prevent this from happening again.

closed 2020-07-09 by whitequark

kgugala, 2020-07-10:
@whitequark looking at this. Shall we reopen this issue?

whitequark, 2020-07-10:
The PRs have been reverted per Claire's requests, so it's no longer a problem in Yosys master. Feel free to resubmit them without the conflicts and we'll merge them again.

kgugala, 2020-07-10:
Thanks, I opened #2257 with fixes

commit dc35ef05 (2020-07-09): verilog_parser: turn S/R and R/R conflicts into hard errors.

Fixes #2253.