YosysHQ/yosys #2258: DFFLEGALIZE pass: ERROR: Conflicting init values for signal 1'0

gregdavill, 2020-07-11:
## Steps to reproduce the issue

I'm building ECP5 SoC's with Litex/migen. Yosys (b822beb1) from 2 weeks is working, but when using Yosys `eef0ec6a` it fails.

I've created a minimal example which produces the same failure: https://github.com/gregdavill/OrangeCrab-examples/blob/minimal/litex/SoC-minimal.py

I have attached the generated files from Litex which are enough to reproduce the issue with just yosys.
[gateware.zip](https://github.com/YosysHQ/yosys/files/4906551/gateware.zip)

This was filed as an issue to me: https://github.com/gregdavill/OrangeCrab-examples/issues/3
The reporter suggests that this recent commit might be to blame: https://github.com/YosysHQ/yosys/commit/372521ca56c2ca27bb562098aa65b4e457959e77

## Expected behavior

When running `yosys orangecrab.ys` I would expect it to complete and produce a `.json` file. 
This is the behaviour of an older version which I was previously using (b822beb1, ~15 days old)

## Actual behavior

Running `yosys orangecrab.ys` results in the following error and terminates.

```
4.39. Executing DFFLEGALIZE pass (convert FFs to types supported by the target).
ERROR: Conflicting init values for signal 1'0 (\interface2_bank_bus_dat_r [2] = 1'x != 1'0).
```

Palakis, 2020-07-11:
I can confirm that I have the same issue with the DFFLEGALIZE pass when running the `build_colorlight_5a_75b.sh` script generated by the `colorlight_5a_75x` target script:

```
4.39. Executing DFFLEGALIZE pass (convert FFs to types supported by the target).
ERROR: Conflicting init values for signal 1'0 (\main_basesoc_interface_adr [6] = 1'x != 1'0).
```

Tested with Yosys eef0ec6a built from git on macOS Catalina 10.15.5:

```
$ yosys --version
Yosys 0.9+2406 (git sha1 eef0ec6a, clang 11.0.3 -fPIC -Os)
```

mwkmwkmwk, 2020-07-12:
Ugh.  This one is not a `dfflegalize` bug, `dfflegalize` just happens to notice the netlist is already subtly broken and refuses to operate.

Consider this example:

```
reg A = 0;
reg B = 1;
always @* A <= B;
always @(posedge clk) B <= something;
```

The init values will of course be stored as `A` and `B` wire attributes. The first process will be optimized to just a connection between these wires, and the second ends up as `$_DFF_P_`. So when something happens to look for the FF's initialization value on all sigmapped wires, it finds both a 0 and a 1. `dfflegalize` considers that an error-worthy situation, of course (as does `zinit`), but many other passes (`dffinit`, `ecp5_ffinit`, ...) just happened to take whichever value was last in the iterator.

(note: the minimal example given above is slightly different since the conflict is between x and 0, and a const signal is involved, but the reason for that is pretty much the same — initial values for things that are not FFs or latches end up in the netlist, instead of getting discarded in `proc`)

Palakis, 2020-07-12:
@mwkmwkmwk So what you're saying is, this is indeed a bug in Yosys but not a bug in the dfflegalize pass?

Ravenslofty, 2020-07-12:
dfflegalize is exposing existing internal bugs, not adding them in this case.

I feel like the error message could be a bit better though.

commit d52a9bd6 (2020-07-12): proc_dlatch: Remove init values for combinatorial processes.

Fixes #2258.
PR #2260 (2020-07-12): proc_dlatch: Remove init values for combinatorial processes.
Fixes #2258.
commit b33744b0 (2020-07-12): proc_dlatch: Remove init values for combinatorial processes.

Fixes #2258.
closed 2020-07-12 by mwkmwkmwk