YosysHQ/yosys #2266: Canot reference items in instantiated module

zachjs, 2020-07-14:
## Steps to reproduce the issue

```Verilog
module Example(a, b);
	input wire [3:0] a;
	output wire [3:0] b;
	assign b = a;
	localparam [3:0] foo = 0;
	localparam [3:0] bar = 1;
endmodule

module top(o1, o2);
	output wire [3:0] o1, o2;
	Example e1(e1.foo, o1);
	Example e2(e2.bar, o2);
endmodule
```

## Expected behavior

`e1.foo` and `e2.bar` should be legal hierarchical references.

## Actual behavior

Warnings are produced, and the ports are left unconnected.
```
hier_bind.v:11: Warning: Identifier `\e1.foo' is implicitly declared.
hier_bind.v:12: Warning: Identifier `\e2.bar' is implicitly declared.
```

whitequark, 2020-07-14:
This works with a script like:
```
hierarchy -auto-top
flatten
# show # uncomment this to see the result
```
It can't work without flattening though.

zachjs, 2020-07-14:
Even with those steps, the resulting ports appear unconnected:
```
"e1.a": {
  "hide_name": 0,
  "bits": [ "x", "0", "0", "0" ],
```
```
assign \e1.a  = 4'b000x;
assign \e1.b  = 4'b000x;
```

Am I missing something?

whitequark, 2020-07-14:
No, you're right, I misread your code.

daveshah1, 2020-07-14:
See #647

widlarizer, 2026-09-14:
Hope I understand correctly that this is a duplicate

closed 2026-09-14 by widlarizer