YosysHQ/yosys #2267: Error when installation directory contains spaces

gregdavill, 2020-07-14:
## Steps to reproduce the issue

On windows install yosys into a directory that has spaces **and** that does not have windows 8.3 filenames enabled.
By default the system drive does have 8.3 files enabled, which mitigates the problem. (As 8.3 filenames do not contain spaces)

You should be able to create drive without 8.3 filenames by formatting an external USB drive as NTFS. 
Running `dir /x` can confirm that 8.3 filenames are not enabled.
```
E:\>dir /x
 Volume in drive E has no label.
 Volume Serial Number is EEF4-B741

 Directory of E:\

14/07/2020  05:36 PM    <DIR>                       blink
14/07/2020  05:08 PM    <DIR>                       fpga toolchain-windows_amd64-nightly-20200712
               0 File(s)              0 bytes
               2 Dir(s)     841,580,544 bytes free
```

Wheras when 8.3 filenames are enbaled you will see this
```
E:\>dir /x
 Volume in drive E has no label.
 Volume Serial Number is EEF4-B741

 Directory of E:\

14/07/2020  05:36 PM    <DIR>          BLINK          blink
14/07/2020  05:08 PM    <DIR>          FPGATO~1       fpga toolchain-windows_amd64-nightly-20200712
               0 File(s)              0 bytes
               2 Dir(s)     841,580,544 bytes free
```

## Expected behavior

Yosys functions normally

## Actual behavior

When running an ecp5 flow `synth_ecp5`, it fails when loading `cells_sim.v` 
```
2.1. Executing Verilog-2005 frontend: E:\fpga toolchain-windows_amd64-nightly-20200712\fpga-toolchain\bin\../share/yosys/ecp5/cells_sim.v
Parsing Verilog input from `E:\fpga toolchain-windows_amd64-nightly-20200712\fpga-toolchain\bin\../share/yosys/ecp5/cells_sim.v' to AST representation.
Generating RTLIL representation for module `\LUT4'.
ERROR: Assert `(unsigned)*c > (unsigned)' '' failed in ./kernel/rtlil.h:168.
make: *** [Makefile:32: blink.json] Error 1
```
The space in the filename here appears to trigger this assertion.

boqwxp, 2020-07-22:
Don't get me wrong, this is a bug and should be fixed, but this is such a commonly occurring kind of bug (especially IMO in FPGA tooling) that I personally think you're a little bit crazy to use spaces in your directory names :).

mmicko, 2026-09-14:
This exact issue is not relevant any more.

closed 2026-09-14 by mmicko