YosysHQ/yosys #2271: write_verilog generate "always @*" blocks that can't be emulated properly

jeanthom, 2020-07-16:
## Steps to reproduce the issue

Consider the following nMigen code:
```python
from nmigen import *
from nmigen.cli import main

class Top(Elaboratable):
	def __init__(self):
		self.a = Signal()

	def elaborate(self, platform):
		m = Module()

		with m.FSM():
			with m.State("aaaaaaaa"):
				m.d.comb += self.a.eq(1)

		# Workaround to
		b = Signal()
		m.d.sync += b.eq(1)

		return m

if __name__ == "__main__":
	top = Top()
	main(top, ports=[top.a])
```

Which will generate the following RTLIL code:
```rtlil
attribute \generator "nMigen"
attribute \top 1
attribute \nmigen.hierarchy "top"
module \top
  attribute \src "verilogfail.py:6"
  wire width 1 output 0 \a
  attribute \src "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/ir.py:526"
  wire width 1 input 1 \clk
  attribute \src "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/ir.py:526"
  wire width 1 input 2 \rst
  attribute \src "verilogfail.py:11"
  wire width 1 \fsm_state
  process $group_0
    assign \a 1'0
    attribute \src "verilogfail.py:11"
    switch \fsm_state
      attribute \src "verilogfail.py:12"
      attribute \nmigen.decoding "aaaaaaaa/0"
      case 1'0
        assign \a 1'1
    end
    sync init
  end
  attribute \src "verilogfail.py:16"
  wire width 1 \b
  attribute \src "verilogfail.py:16"
  wire width 1 \b$next
  process $group_1
    assign \b$next \b
    assign \b$next 1'1
    attribute \src "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/xfrm.py:519"
    switch \rst
      case 1'1
        assign \b$next 1'0
    end
    sync init
      update \b 1'0
    sync posedge \clk
      update \b \b$next
  end
  connect \fsm_state 1'0
end
```

Which will result in the following Verilog:
```verilog
/* Generated by Yosys 0.9+2406 (git sha1 021ce8e5, gcc 10.1.1 -fPIC -Os) */

(* \nmigen.hierarchy  = "top" *)
(* top =  1  *)
(* generator = "nMigen" *)
module top(clk, rst, a);
  (* src = "verilogfail.py:6" *)
  output a;
  reg a;
  (* src = "verilogfail.py:16" *)
  reg b = 1'h0;
  (* src = "verilogfail.py:16" *)
  reg \b$next ;
  (* src = "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/ir.py:526" *)
  input clk;
  (* src = "verilogfail.py:11" *)
  wire fsm_state;
  (* src = "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/ir.py:526" *)
  input rst;
  always @(posedge clk)
      b <= \b$next ;
  always @* begin
    a = 1'h0;
    (* src = "verilogfail.py:11" *)
    casez (fsm_state)
      /* \nmigen.decoding  = "aaaaaaaa/0" */
      /* src = "verilogfail.py:12" */
      1'h0:
          a = 1'h1;
    endcase
  end
  always @* begin
    \b$next  = 1'h1;
    (* src = "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/xfrm.py:519" *)
    casez (rst)
      1'h1:
          \b$next  = 1'h0;
    endcase
  end
  assign fsm_state = 1'h0;
endmodule
```

## Expected behavior

When the above Verilog code is emulated using a Verilog simulator (say Icarus Verilog), `a` register isn't initialized because there is no event at t=0 triggering the always @* block.

## Actual behavior

At t=0, `a` is set to its initial value.

## Related

nMigen issue for this bug: https://github.com/nmigen/nmigen/issues/418 (cc @whitequark)

whitequark, 2020-07-16:
For context, we already discussed this on a Yosys meeting, and decided that the solution here differs by the mode. In SystemVerilog mode (which doesn't exist yet), `write_verilog` should just use `always_comb`. In Verilog-2001 mode, something like this:
```verilog
reg init;
initial init = 0;
reg x;
always @* begin
    if (init) begin end
    x <= 1;
end
```

whitequark, 2020-07-16:
See #2272 and #2273.

closed 2020-07-16 by clairexen

whitequark, 2020-08-26:
Reopening because Verilog-2001 mode is still broken.

commit 128522f1 (2020-07-16): verilog_backend: in non-SV mode, add a trigger for `always @*`.

This commit only affects translation of RTLIL processes (for which
there is limited support).

Due to the event-driven nature of Verilog, processes like

    reg x;
    always @*
        x <= 1;

may never execute. This can be fixed in SystemVerilog code by using
`always_comb` instead of `always @*`, but in Verilog-2001 the options
are limited. This commit implements the following workaround:

    reg init = 0;
    reg x;
    always @* begin
        if (init) begin end
        x <= 1;
    end

Fixes #2271.