YosysHQ/yosys #2279: Yosys only instantiates bram correctly if it is read into a verilog output register

Niautanor, 2020-07-19:
## Steps to reproduce the issue

When I try to synthesize the following code with yosys (for ICE40 but I think that isn't relevant here), it does not implement `memory` as a bram. During the MEMORY_BRAM pass, it remarks that the read port is in the clock domain `~!async!`.

```
`default_nettype none

module vga(input clk, output [1:0] r, output [1:0] g, output [1:0] b);
reg [7:0] pixel;

reg [9:0] x = 0;
wire [9:0] x_next = x + 1;
always @(posedge clk) x <= x_next;

reg [9:0] y = 0;
wire [9:0] y_next = y + 1;
always @(posedge clk) y <= y_next;

reg [7:0] memory[0:15][0:15];
initial $readmemh("data/image.hex", memory);

always @(posedge clk) begin
    pixel <= memory[x_next][y_next];
end

assign r = pixel[5:4];
assign g = pixel[3:2];
assign b = pixel[1:0];

endmodule
```

When I turn `pixel` from a local register into a module output, `memory` gets correctly turned into a bram.

```
`default_nettype none

module vga(input clk, output reg [7:0] pixel, output [1:0] r, output [1:0] g, output [1:0] b);

reg [9:0] x = 0;
wire [9:0] x_next = x + 1;
always @(posedge clk) x <= x_next;

reg [9:0] y = 0;
wire [9:0] y_next = y + 1;
always @(posedge clk) y <= y_next;

reg [7:0] memory[0:15][0:15];
initial $readmemh("data/image.hex", memory);

always @(posedge clk) begin
    pixel <= memory[x_next][y_next];
end

assign r = pixel[5:4];
assign g = pixel[3:2];
assign b = pixel[1:0];

endmodule
```

## Notes

Running synthesis until the MEMORY_BRAM pass and dumping the results of that gives me this cell for my memory

```
  cell $mem \memory
    parameter \ABITS 14
    parameter \INIT 131072'// snip
    parameter \MEMID "\\memory"
    parameter \OFFSET 0
    parameter \RD_CLK_ENABLE 1'0
    parameter \RD_CLK_POLARITY 1'0
    parameter \RD_PORTS 1
    parameter \RD_TRANSPARENT 1'0
    parameter \SIZE 16384
    parameter \WIDTH 8
    parameter \WR_CLK_ENABLE 1'0
    parameter \WR_CLK_POLARITY 1'0
    parameter \WR_PORTS 0
    connect \RD_ADDR $add$vga.v:0$24_Y
    connect \RD_CLK 1'x
    connect \RD_DATA $0\pixel[7:0]
    connect \RD_EN 1'x
    connect \WR_ADDR { }
    connect \WR_CLK { }
    connect \WR_DATA { }
    connect \WR_EN { }
  end
```

Note the `1'x` for `\RD_CLK` and `\RD_EN`. I thought that was odd. For the working example, the values connected there are `\clk` and `1'1` respectively.

daveshah1, 2020-07-19:
I believe this is a bug where the unused bits mean the read register isn't read correctly - see #1854 for a smaller testcase that exhibits the same problem.

Niautanor, 2020-07-19:
Changing the memory to be only 6 bit wide fixes the issue so you are totally right that this is a duplicate.

closed 2020-07-19 by Niautanor