YosysHQ/yosys #2280: Verilog2001 style parameters

stephenry, 2020-07-19:
The following does not appear to be supported:

```systemverilog
module foo #(parameter int W) (input [W-1:0] x,
			       output logic [W-1:0] y);
   always_comb
     y = ~x;

endmodule // foo
```
Unfortunately, I have a relatively large codebase that makes use of this idiom. Is there a workaround for it?

daveshah1, 2020-07-19:
There are two problems here, neither of which are the Verilog-2001 parameter style.

1. the SystemVerilog `int` type is not supported (I think for Yosys purposes this can be an alias of `integer` which is supported, so not the hardest of fixes)
2. parameters without a default value are not supported, another SystemVerilog addition

I can't offer a very useful workaroud, but the following code works fine for reference:

```systemverilog
module foo #(parameter W = 0) (input [W-1:0] x,
			       output logic [W-1:0] y);
   always_comb
     y = ~x;

endmodule
```

stephenry, 2020-07-19:
Thank you for your assistance. Your suggested changes do appear to resolve the issue, but I run into new issues soon thereafter (automatic storage, multi-dimensional array of a type). The code is quite heavy SystemVerilog, so probably a lot of it is unsupported at the moment.

eddiehung, 2020-07-19:
Are you aware of https://github.com/zachjs/sv2v?

stephenry, 2020-07-19:
Gave this a quick look. May indeed be a very usable approach. Run into problem where Yosys complains a "_non-constant function call in constant expression_"; where all arguments are localparam. The cast function generated by sv2v appears to be constant to me (although I could be wrong).

```systemverilog
        localparam signed [31:0] pa_pkg_SPaDescBytesN = 5;
	function automatic signed [31:0] sv2v_cast_32_signed;
		input reg signed [31:0] inp;
		sv2v_cast_32_signed = inp;
	endfunction
	localparam signed [31:0] pa_stack_pkg_DescPerLineN = sv2v_cast_32_signed($floor(pa_stack_pkg_LineBytesN / pa_pkg_SPaDescBytesN));
```

zachjs, 2020-07-20:
@stephenry I opened https://github.com/YosysHQ/yosys/pull/2281 which allows real values (such as those produced by `$floor`) as constant function arguments.

It's worth noting sv2v doesn't yet support parameters without default values. I've wanted to add support for some time, but I'm curious if tossing in some arbitrary default, say `0`, would be a reasonable conversion.

zachjs, 2021-03-07:
I believe all of the features specifically mentioned in this issue have now bin fixed. On master, the original example appears to behave as expected.

closed 2021-03-07 by zachjs