YosysHQ/yosys #2290: Crash when attempting to read verilog

stephenry, 2020-07-21:
The attached file has some procedurally generated Verilog (from SystemVerilog using SV2V). When attempting to read the verilog into Yosys, the tool appears to crash without an error message. I gave the RTL a quick lookover and it appears to be wellformed so therefore unsure about how to debug.

[crash.tar.gz](https://github.com/YosysHQ/yosys/files/4956777/crash.tar.gz)

Ravenslofty, 2020-07-22:
Here's the ASAN splat:
```
1. Executing Verilog-2005 frontend: pa_stack_pipe_fwd_queue.v
Parsing Verilog input from `pa_stack_pipe_fwd_queue.v' to AST representation.
Generating RTLIL representation for module `\pa_stack_pipe_fwd_queue'.
=================================================================
==7017==ERROR: AddressSanitizer: allocator is out of memory trying to allocate 0xf6600000 bytes
    #0 0x55606ee465d9 in operator new(unsigned long) (/home/lofty/yosys/yosys+0xcea5d9)
    #1 0x55606ef6c56e in __gnu_cxx::new_allocator<Yosys::RTLIL::SigBit>::allocate(unsigned long, void const*) /usr/sbin/../lib64/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../include/c++/10.1.0/ext/new_allocator.h:115:27
    #2 0x55606ef6c54a in std::allocator_traits<std::allocator<Yosys::RTLIL::SigBit> >::allocate(std::allocator<Yosys::RTLIL::SigBit>&, unsigned long) /usr/sbin/../lib64/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../include/c++/10.1.0/bits/alloc_traits.h:460:20
    #3 0x55606ef6c52d in std::_Vector_base<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::_M_allocate(unsigned long) /usr/sbin/../lib64/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../include/c++/10.1.0/bits/stl_vector.h:346:20
    #4 0x55606efc1e08 in void std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::_M_range_insert<__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > > >(__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, __gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, __gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, std::forward_iterator_tag) /usr/sbin/../lib64/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../include/c++/10.1.0/bits/vector.tcc:769:29
    #5 0x55606efc1b8f in void std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::_M_insert_dispatch<__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > > >(__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, __gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, __gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, std::__false_type) /usr/sbin/../lib64/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../include/c++/10.1.0/bits/stl_vector.h:1665:4
    #6 0x55606ef5aafa in __gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > > std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::insert<__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, void>(__gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, __gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >, __gnu_cxx::__normal_iterator<Yosys::RTLIL::SigBit const*, std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> > >) /usr/sbin/../lib64/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../include/c++/10.1.0/bits/stl_vector.h:1383:4
    #7 0x55606ef2db63 in Yosys::RTLIL::SigSpec::append(Yosys::RTLIL::SigSpec const&) /home/lofty/yosys/kernel/rtlil.cc:3873:9
    #8 0x55606f202cab in Yosys::AST::AstNode::genRTLIL(int, bool) /home/lofty/yosys/frontends/ast/genrtlil.cc:1335:9
    #9 0x55606f1ffea0 in Yosys::AST::AstNode::genRTLIL(int, bool) /home/lofty/yosys/frontends/ast/genrtlil.cc:1320:23
    #10 0x55606f202c67 in Yosys::AST::AstNode::genRTLIL(int, bool) /home/lofty/yosys/frontends/ast/genrtlil.cc:1329:40
    #11 0x55606f1ffea0 in Yosys::AST::AstNode::genRTLIL(int, bool) /home/lofty/yosys/frontends/ast/genrtlil.cc:1320:23
    #12 0x55606f202c67 in Yosys::AST::AstNode::genRTLIL(int, bool) /home/lofty/yosys/frontends/ast/genrtlil.cc:1329:40
    #13 0x55606f212d04 in Yosys::AST::AstNode::genWidthRTLIL(int, Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::SigBit, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> > const*) /home/lofty/yosys/frontends/ast/genrtlil.cc:1879:23
    #14 0x55606f218f31 in Yosys::AST_INTERNAL::ProcessGenerator::processAst(Yosys::AST::AstNode*) /home/lofty/yosys/frontends/ast/genrtlil.cc:562:47
    #15 0x55606f219637 in Yosys::AST_INTERNAL::ProcessGenerator::processAst(Yosys::AST::AstNode*) /home/lofty/yosys/frontends/ast/genrtlil.cc:555:5
    #16 0x55606f219637 in Yosys::AST_INTERNAL::ProcessGenerator::processAst(Yosys::AST::AstNode*) /home/lofty/yosys/frontends/ast/genrtlil.cc:555:5
    #17 0x55606f214ef9 in Yosys::AST_INTERNAL::ProcessGenerator::ProcessGenerator(Yosys::AST::AstNode*, Yosys::RTLIL::SigSpec) /home/lofty/yosys/frontends/ast/genrtlil.cc:396:5
    #18 0x55606f1fea45 in Yosys::AST::AstNode::genRTLIL(int, bool) /home/lofty/yosys/frontends/ast/genrtlil.cc:1772:21
    #19 0x55606f18ec66 in Yosys::process_module(Yosys::AST::AstNode*, bool, Yosys::AST::AstNode*, bool) /home/lofty/yosys/frontends/ast/ast.cc:1119:11
[... cut ...] x11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, Yosys::RTLIL::Design*) /home/lofty/yosys/kernel/yosys.cc:988:6
    #26 0x55606ee4b30a in main /home/lofty/yosys/kernel/driver.cc:532:4
    #27 0x7fb0fcb05001 in __libc_start_main (/usr/lib/libc.so.6+0x27001)

==7017==HINT: if you don't care about these errors you may set allocator_may_return_null=1
SUMMARY: AddressSanitizer: out-of-memory (/home/lofty/yosys/yosys+0xcea5d9) in operator new(unsigned long)
==7017==ABORTING
```

Ravenslofty, 2020-07-22:
After talking with @mwkmwkmwk we think the problem is this line:
```verilog
p3_queue_w = {((QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1)) >= (QueueN >= 1 ? 0 : (QueueN - 1) * 2) ? ((QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1)) - (QueueN >= 1 ? 0 : (QueueN - 1) * 2)) + 1 : ((QueueN >= 1 ? 0 : (QueueN - 1) * 2) - (QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1))) + 1) {{((QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1)) >= (QueueN >= 1 ? 0 : (QueueN - 1) * 2) ? ((QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1)) - (QueueN >= 1 ? 0 : (QueueN - 1) * 2)) + 1 : ((QueueN >= 1 ? 0 : (QueueN - 1) * 2) - (QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1))) + 1) {{((QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1)) >= (QueueN >= 1 ? 0 : (QueueN - 1) * 2) ? ((QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1)) - (QueueN >= 1 ? 0 : (QueueN - 1) * 2)) + 1 : ((QueueN >= 1 ? 0 : (QueueN - 1) * 2) - (QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1))) + 1) {{((QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1)) >= (QueueN >= 1 ? 0 : (QueueN - 1) * 2) ? ((QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1)) - (QueueN >= 1 ? 0 : (QueueN - 1) * 2)) + 1 : ((QueueN >= 1 ? 0 : (QueueN - 1) * 2) - (QueueN >= 1 ? (QueueN * 2) - 1 : ((2 - QueueN) * 2) + (((QueueN - 1) * 2) - 1))) + 1) {{(QueueN >= 1 ? QueueN : 2 - QueueN) {{2 {p3_queue_r}}}}}}}}}}}};
```

which reduces to

```verilog
p3_queue_w = {48 {{48 {{48 {{48 {{24 {{2 {p3_queue_r}}}}}}}}}}}};
```

or a 254,803,968-bit intermediate expression, of which you only need the lower 48 bits.

I'm inclined to call this a bug in sv2v and suggest you file a bug report there; this code seems *very wrong*.

whitequark, 2020-07-22:
I believe this is still an issue in Yosys, since it ought to reject things it *definitely* cannot synthesize. In my experience, anything over a million bits is clearly wrong, and that doesn't even lead to OOM, just to a silent hang.

whitequark, 2020-07-22:
Here's an issue report I got just yesterday: https://twitter.com/dvdfreitag/status/1285730827272429570

Ravenslofty, 2020-07-22:
Then there are two bugs: one in sv2v for producing this monstrous wire, and another in Yosys for not cleanly handling it.

whitequark, 2020-07-22:
Agreed. Let's fix this in Yosys!

Ravenslofty, 2020-07-22:
I think we should also cc @zachjs because of that; hopefully @stephenry can provide the input SV code to help debug it.

Ravenslofty, 2020-07-22:
It turns out I got my math wrong; the expression is actually almost 1.8 billion bits wide.

stephenry, 2020-07-22:
Folks, the original SV and ancilliary headerfiles can be found at:

[fail.tar.gz](https://github.com/YosysHQ/yosys/files/4962048/fail.tar.gz)

zachjs, 2020-07-22:
The issue in sv2v causing those needless repeats was fixed yesterday evening. That particular output line is now simply:
```Verilog
p3_queue_w = p3_queue_r;
```

stephenry, 2020-07-22:
I can confirm on my end that the generated Verilog is now successfully parsed with the new fix to SV2V.

PR #2632 (2021-03-07): verilog: impose limit on maximum expression width
Designs with unreasonably wide expressions would previously get stuck allocating memory forever.

Fixes #2290
closed 2021-03-07 by whitequark