YosysHQ/yosys #2296: Constant arg loop elaboration in function with some non-constant args

zachjs, 2020-07-24:
## Steps to reproduce the issue

```Verilog
module top;
	function automatic [31:0] operation;
		input reg [3:0] rounds;
		input integer x;
		integer i;
		begin
			for (i = 0; i < rounds[3:0]; i = i + 1)
				x = x * 2;
			operation = x;
		end
	endfunction

	wire a;
	assign a = 2;

	parameter A = 3;

	wire [31:0] x;
	assign x = operation(A, a);

// `define VERIFY
`ifdef VERIFY
    assert property (a == 2);
    assert property (A == 3);
    assert property (x == 16);
`endif
endmodule
```

## Expected behavior

The loop in `operation` should be successfully elaborated given that is invoked with `rounds` as a constant.

## Actual behavior

```
test.v:7: ERROR: 2nd expression of generate for-loop is not constant!
```

Ravenslofty, 2020-07-24:
As I read the Verilog-2005 standard, Yosys is correct.

> The operands of a constant
expression consist of constant numbers, strings, parameters, constant bit-selects and part-selects of parameters, constant function calls (see 10.4.5), and constant system function calls only

and

> The data types reg, integer, time, real, and realtime are all variable data types.

`rounds[3:0]` is not a constant expression because `rounds` is `reg`, which is a variable and not a `parameter`.

zachjs, 2020-07-25:
The invocation of `operation()` in this example is not a `constant_function_call`, as the right hand side of a continuous assignment is not required to be a constant expression.

Yosys should be able to unroll the loop in this function given that `rounds` is given a constant expression, and is itself never procedurally assigned within the function. I've poked around a bit further and I think I can add this functionality in a PR soon.

clairexen, 2020-07-26:
According to IEEE Std. 1364.1 a `constant_expression` is an expression of `constant_primary`, which in trurn is a `genvar_identifier`, `number`, `parameter_identifier` or `specparam_identifier`. The `rounds` identifier above however is an `hierarchical_identifier` that is not a `genvar_identifier`, `parameter_identifier` or `specparam_identifier`. Thus this is a `primary` and the whole thing an `expression`, not a `constant_primary` and a `constant_expression`.

Ravenslofty, 2020-07-26:
Okay, so Yosys is in the wrong because `rounds` is not required to be a `constant_expression` in this context?

clairexen, 2020-07-26:
> Okay, so Yosys is in the wrong because `rounds` is not required to be a `constant_expression` in this context?

Well.. IEEE Std. 1364.1-2002 says in 7.7.6

> Loop bounds shall be statically computable for a for loop

and 3.15

> statically computable expression: An expression whose value can be evaluated during compilation.

However, in the Yosys `read_verilog` front-end we only support constant_expressions as statically computable expressions.

Supporting IEEE Std. 1364.1-2002 to the full extend in this regard would require a complete rewrite of the front-end. (However, I'm sure no tool out there supports that to the full extend. In fact, I had to change designs to make them compatible with other tools because in some cases Yosys actually has better support for statically computable expressions than some other tools.)

zachjs, 2020-07-26:
I don't think Yosys should be expected to employ more advanced analysis in attempting to unroll loops. The strategy I use in my PR to work around this is to turn function arguments into localparams (rather than wires) where trivially possible. Hopefully, this expands the capabilities of Yosys without compromise.

clairexen, 2020-07-26:
> turn function arguments into localparams (rather than wires) where trivially possible

I think that's a good way to do it. :+1:

PR #2299 (2020-07-26): Avoid generating wires for function args which are constant
Resolves #2296. Any feedback is greatly appreciated!
closed 2020-07-26 by clairexen