YosysHQ/yosys #2310: Feature request: $display cell

jeanthom, 2020-07-28:
A few persons (including myself) have expressed the need for a $display cell that would allow to write things on a console from inside a module (like $display in Verilog). This would make debugging far easier.

Related issue in nMigen: https://github.com/nmigen/nmigen/issues/432

mwkmwkmwk, 2020-07-28:
Could you tell us more about the cell you'd want?  What would be the inputs and semantics?  Which parts of yosys should support it?  (I'd guess Verilog frontend, cxxrtl, Verilog backend?)

whitequark, 2020-07-28:
@mwkmwkmwk Didn't we discuss it in a meeting a few weeks ago?

whitequark, 2020-07-28:
I believe the semantics we worked out was along the lines of:
```verilog
module $display(...);
    parameter CLK_ENABLE = 1'b0;
    parameter CLK_POLARITY = 1'b0;
    parameter FORMAT_WIDTH;
    parameter ARGS_WIDTH;
    parameter ARGS_LAYOUT;
    input CLK;
    input EN;
    input [FORMAT_WIDTH:0] FORMAT;
    input [ARGS_WIDTH:0] ARGS;
endmodule
```

The idea is to have a primitive that maps well to Verilog backend, CXXRTL backend, and formal verification. The function of `CLK` and `EN` is the same as elsewhere. `ARGS` is all of the substitutions, concatenated together. `ARGS_LAYOUT` is a list of argument sizes pre-concatenation, to make the argument list unambiguous without any inspection of the format strings.

Synchronous display cells trigger on the clock. Asynchronous display cells trigger on any change of inputs. (This was Claire's suggestion.)

mwkmwkmwk, 2020-07-28:
Yeah, sounds good (I just haven't seen it written down before).

daglem, 2023-03-29:
If I understand correctly, this is handled in #2459.

povik, 2023-09-28:
This feature has been added under the name of `$print` cells in #3721.

closed 2023-09-28 by povik